solosky/pixl.js #82: 请问官方版pixl.ja可以线刷这个吗？

czh00, 2023-06-03:
买了官方版ota不能刷，用线刷的可以用吗？怕刷成砖。

solosky, 2023-06-04:
可以

czh00, 2023-06-10:
用PWLINK2接的缐刷完显示成功但是不能开机了，这是砖了吗？
有无方法挽救？

solosky, 2023-06-10:
是下载最新版本固件，然后使用fw_update.bat脚本更新的吗？
n能否提供更多信息？比如截图，错误信息等。

czh00, 2023-06-10:
是下载2.6.0用fw_update.bat更新的，最后显示 "恭喜，刷新固件成功！" 没任何错误信息

solosky, 2023-06-10:
如果你是在咸鱼买的，建议找卖家售后吧

czh00, 2023-06-10:
------------------------------------------------------------------------------
Pixl.js 固件自動燒寫腳本 @solosky
------------------------------------------------------------------------------
openocd已經安裝，跳過下載
檢查openocd版本
Open On-Chip Debugger 0.12.0 (2023-01-14-23:37)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
刷新固件中，請稍後...
Open On-Chip Debugger 0.12.0 (2023-01-14-23:37)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: FW Version = 1.11
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 1000 kHz
Info : SWD DPIDR 0x2ba01477
Info : [nrf52.cpu] Cortex-M4 r0p1 processor detected
Info : [nrf52.cpu] target has 6 breakpoints, 4 watchpoints
Info : starting gdb server for nrf52.cpu on 3333
Info : Listening on port 3333 for gdb connections
[nrf52.cpu] halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x00000a80 msp: 0x20000400
Info : nRF52832-QFAA(build code: E1) 512kB Flash, 64kB RAM
Info : Mass erase completed.
[nrf52.cpu] halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
** Programming Started **
Info : Padding image section 0 at 0x00000b00 with 1280 bytes
Info : Padding image section 1 at 0x00018fec with 20 bytes
Info : Flash write discontinued at 0x00068930, next section at 0x00077000
Warn : Adding extra erase range, 0x00068930 .. 0x00068fff
Info : Padding image section 3 at 0x0007dc98 with 872 bytes
Info : Padding image section 4 at 0x0007e05c with 4004 bytes
Warn : Adding extra erase range, 0x0007f05c .. 0x0007ffff
Warn : Adding extra erase range, 0x10001000 .. 0x10001013
Warn : Adding extra erase range, 0x1000101c .. 0x10001fff
** Programming Finished **
** Verify Started **
** Verified OK **
-----------------------------------------------------------------------------
恭喜，刷新固件成功！
-----------------------------------------------------------------------------
請按任意鍵繼續 . . .

czh00, 2023-06-10:
espruinoespruino官方买的售后是不可能了我再找看看是否有其他方法能解决

solosky, 2023-06-10:
你是说买了这个：https://shop.espruino.com/pixljs？
如果是这个话，那和我这个固件不兼容。

您可能需要从 https://github.com/espruino/Espruino clone源码自行编译原版的Pixl.js的固件，然后通过线刷恢复。

czh00, 2023-06-10:
看来是没死透swclk/swdio接上去都还能刷只是画面不显示，正在想办法刷回官方固件。

solosky, 2023-06-10:
[espruino_2v18.25_pixljs.zip](https://github.com/solosky/pixl.js/files/11711838/espruino_2v18.25_pixljs.zip)
你试试下载这个文件解压出来espruino_2v18.25_pixljs.hex， 然后重命名为pixljs_all.hex，替换掉固件包里面的pixljs_all.hex，然后执行fw_update.bat刷新到设备试试。

czh00, 2023-06-10:
刷回官方固件了感谢支援
用你的韧体盲试了一下功能都有，似乎只是显示不匹配？

solosky, 2023-06-10:
我的固件需要适配下官方的屏幕，但因为官方没有flash，所以只能使用amiibolink模式

closed 2023-06-13 by solosky