solosky/pixl.js #84: Language change

Pocketmodster, 2023-06-12:
Hello, how do I change the language on the device to English? Thank you

Bosco1262, 2023-06-13:
I think you need to use another version

solosky, 2023-06-13:
Sorry, Pixl.js doesn't have any other language translation for both firmware and web, it only supports Simplified Chinese. 

I will leave the issue open and may schedule English language support in future, but I am not promise it. 

Also, pull request is welcome if you are capable of translation for English, includes firmware, web and documentations.

headhunter45, 2023-06-15:
It doesn't look like the code is setup for l10n yet. Do you have a plan or preference for how to set that up? I just started digging in, but I think I could setup something with a compile time define. I haven't looked too deep into how the settings work, but I can think of a few approaches for runtime language switching. I'm not sure on the web stuff though, maybe something similar, but I've been out of web dev for a few years now.

solosky, 2023-08-07:
Since version 2.7.0 is offically support English language now, close this issue as completed. 
See #92 .

closed 2023-08-07 by solosky