solosky/pixl.js #90: 无效数据文件导致唤醒后卡死

x-magic, 2023-06-17:
我有几个bin dump在pixl.js显示“这不是Amiibo文件”，但使用amiitool测试通过：
```bash
./amiitool -k keys.bin -i original.bin -d
```
可以正常加解密：
```bash
./amiitool -k keys.bin -i original.bin -d -o decrypted.bin
./amiitool -k keys.bin -i decrypted.bin -e -o reencrypted.bin
```
加密后的dump就可以被正确识别了

不过这不是大问题 - 问题是如果在“这不是Amiibo文件”界面时设备进入休眠，唤醒后（快速唤醒开启）错误界面会一闪而过，然后就白屏死机了，必须抠电池才能恢复。感觉和 #40 可能有点关系。

可以使用下面的参考文件 `original.bin` 在 v2.6.2 上复现。测试硬件上除了SPI Flash是32Mbit以外，应该没有什么特殊的。

文件供参考：
```
original.bin (base64)
BDTCehq6TIBsSA/g8RD/7qUAAACENrfcdPVJ8bNW/bZprfvf5dQuNXk/70C60vWCf8lnLja5JC35IoQOUB8qQpM30Nso9pPWG08Fpu4h6jZO4f27MkAAAAJbAAINEicRQ0WqHp7SkzlFIPMVGBofZCTpwCAWuRMVA9HHrYyBuZ5adBfC+TAnryRplfyWqElehralUViMLKoLWGl9oZAmBQVfHRKvIas1QicQoxDrNzj8S9iwRKZZ1RI2qD6MTvA6LS4gNBmprNG0NxDvSVxKIyg5/TajWmrFpQWnQK4JXwmWC6RrA+/d4x6QMWimfBupvyoKHXH/X3zv5ASidLzmyiT8YdrWYRBJXpHJ3Ytz//Bxh9s/6GyumVuGJ/2srPKytxv1Spl3VEtTgRMIe3C/YNb/lFmZR3y5HqjX7N0pdtXzRzfz2H40jsZ0/6F1VaFdke+9TfY8xVc0O3+fGIw118zZuksF0mx9fY8qathw8sRSzPUyvzqZcMwNvR4k9uhyzV+hvYQ0exqQ5YDAiSkruIASwutnHG7xPMFtddTgThMoN2HTSw494cXUPVQVx1OhtdldtrZJFJBObQ8nvvGU3yhl8LXBbvEubczCpfgxwzyZdV/x/R1fJPlF/X2DVke2J/1eX0CwSNEpsKYmnD5swM2Q79lnk9bvMnmfYWqgtJHc2c2Y0lYgCgEAD70AAAAEXwAAAAAAAAAAAAAAZoC5Mn66IQ3JUvCf3KJdtx5vWgvuEkesTPVVmnYmQXY=

decrypted.bin (base64)
bEgP4PEQ/+5adBfC+TAnryRplfyWqElehralUViMLKoLWGl9oZAmBaUAAAAAAAAAAtR1BAI8KGoJthUx83y6afm09NHPzIjQwnU7kTZA3Z00sYg+Sv9anTTdqFHqaf8KSxJ0XN6/ybCYeJIp2iBddk6iWnAC9AGJttySK9UwCBKBgd0ky4C+defYpdFtMcMUfzIDRw6PCFcR66YCLZ7baHB/qFhCOarO5OWIKk8wHIUYfayjAAAtwATxeYYT4Nm7R0n9PWDtiGvSRoc0YygH3gfgzyHqr/I3OrB7tt+7Jsn0KJ0/LBHPLvytq2Ub+F3b6pt/MhG40ZhXxQOwcdGLFyC+QDqLx7H2EaAkPQSMHEFToMxDPPO4PFcF52cOHr0rIZRFXbgiUCLpiSFxk5SUy6jBXdEFR9osBQTjhd9J5/Lj+RtwRr+JMVCpWbPSR6jDcRy+TMHsJtm5GVezPspCoKEujZp4I4TpOCyAW+0nWb5cHxTDi5ow99NBti1n1fKGZrkRo7kOS02X1RSVa8rrkr+YVx+2fxQG/pbGNm1XgWM5+vQ0FbYFTdTGK7aGm4sIViRtzja5JC35IoQOUB8qQpM30Nso9pPWG08Fpu4h6jZO4f27BDTCehq6TIAyQAAAAlsAAg0SJxFDRaoentKTOUUg8xUYGh9kJOnAIBa5ExUD0cetjIG5ngEAD70AAAAEXwAAAAAAAAAAAAAA

reencrypted.bin (baset64)
BDTCehq6TIBsSA/g8RD/7qUAAACENrfcdPVJ8bNW/bZprfvf5dQuNXk/70C60vWCf8lnLja5JC35IoQOUB8qQpM30Nso9pPWG08Fpu4h6jZO4f27MkAAAAJbAAINEicRQ0WqHp7SkzlFIPMVGBofZCTpwCAWuRMVA9HHrYyBuZ5adBfC+TAnryRplfyWqElehralUViMLKoLWGl9oZAmBQVfHRKvIas1QicQoxDrNzj8S9iwRKZZ1RI2qD6MTvA6LS4gNBmprNG0NxDvSVxKIyg5/TajWmrFpQWnQK4JXwmWC6RrA+/d4x6QMWimfBupvyoKHXH/X3zv5ASidLzmyiT8YdrWYRBJXpHJ3Ytz//Bxh9s/6GyumVuGJ/2srPKytxv1Spl3VEtTgRMIe3C/YNb/lFmZR3y5HqjX7N0pdtXzRzfz2H40jsZ0/6F1VaFdke+9TfY8xVc0O3+fGIw118zZuksF0mx9fY8qathw8sRSzPUyvzqZcMwNvR4k9uhyzV+hvYQ0exqQ5YDAiSkruIASwutnHG7xPMFtddTgThMoN2HTSw494cXUPVQVx1OhtdldtrZJFJBObQ8nvvGU3yhl8LXBbvEubczCpfgxwzyZdV/x/R1fJPlF/X2DVke2J/1eX0CwSNEpsKYmnD5swM2Q79lnk9bvMnmfYWqgtJHc2c2Y0lYgCgEAD70AAAAEXwAAAAAAAAAAAAAA
```

solosky, 2023-06-17:
感谢反馈，有意思的BUG，我回家测试下先。

x-magic, 2023-06-17:
更新一下，导致错误的bin比可以使用的bin大了32B。还没来得及看Amiibo dump的结构，难不成是数据部分？（但也不应该啊，毕竟NTAG215也只能写540B）前面的540B其实是一样的。

ca1e, 2023-06-19:
其原因是有一些dump软件在保存数据时会添加/省略一些额外的配置字节。这些配置字节在正常ntag21x卡片中提供功能，但在amiibo识别中被忽略。amiitool工具在加解密时不在乎这些数据，pixljs在模拟时也会忽略这些数据从而可以更简单的实现amiibo的功能。

标准的ntag215卡片的dump数据大小就是540字节，这是毋庸置疑的。其他'非标'大小(532/572)的数据完全可以通过手动补'0'/截断来转换成标准dump数据，这对模拟amiibo没有任何影响。并且我也并不建议对这些'非标'文件进行兼容，要解决这个问题也很简单，在上传/接受文件时判断一下size就行

另外，目前bin的来源可以通过https://github.com/ca1e/amiitool.net 这个项目的`-g`命令来生成全部数据，在pixljs的v3分支也在实现内置生成数据的功能，可以说以后在模拟流程中bin数据其实已经不再重要了。

x-magic, 2023-06-28:
> 在pixljs的v3分支也在实现内置生成数据的功能，可以说以后在模拟流程中bin数据其实已经不再重要了。

这个蛮有意思的，这么说的话验证逻辑完全可以做进客户端（iOS/Android/Web），而不是服务端（pixl.js）。那么如果假设写入pixl.js的数据一定是有效的（540字节），睡眠死机的问题还需要debug吗？

commit c0336b3b (2023-09-07): add debug make file configuration and fixed #90
closed 2023-09-09 by solosky