solosky/pixl.js #111: Putting files inside a subfolder makes the subfoler into a zip?????

Billybobertjoe, 2023-08-30:
[image]

solosky, 2023-08-31:
Hi sir, 

Please provide more information here, for example,

* Detail steps to produce this issue 
* What is expected behavior 
* What is ectual behavior 

it is even better if you post screenshots or demo videos here.

Billybobertjoe, 2023-09-02:
I go to iNFC app and connect to the pixl.js and create a folder inside the amiibo folder. I then create another folder inside that. I then create another folder inside that. I then place the FOLDER containing the amiibo bin files inside that folder. That usually causes the pixl.js to make the FOLDER into a zip?

solosky, 2023-09-02:
Actually, I dont understand 'make the FOLDER into a zip', can you explain more about this ?

Billybobertjoe, 2023-09-02:
> [image]

Like this.

Billybobertjoe, 2023-09-02:
The "amiibo figures" file is supposed to be a folder.

Billybobertjoe, 2023-09-02:
It is turning it into a file of some kind instead of keeping it as a folder

solosky, 2023-09-02:
Oh, I get your point, the fact is that the Pixl.js file system has limitation on its full file path for a file, like /amiibo/animalcrossing/amiibo1.bin. The total length of the file should not be exceed 64 characters, and file/folder name should not be exceed 48 characters. This limitation is introduced by SPIFFS file system. 
```
#define VFS_MAX_NAME_LEN 48
#define VFS_MAX_PATH_LEN 64
#define VFS_MAX_META_LEN 128
```

Please shorten your folder name and file name, and keep the folder count minimum as you can. 

Meanwhile, you can use iNFC update to latest 2.8.3, it introduce a new app called Amiibo Database with almost all amiibo files built in, you can have a try. 

Please let me know if you have more questions.

solosky, 2023-09-02:
I think I will impove the error prompt in future release, sorry to make your confused.

Billybobertjoe, 2023-09-02:
Ok thank youSent from my iPhone

solosky, 2023-10-19:
Close the issue since there is no furthermore comments from user.

closed 2023-10-19 by solosky

Billybobertjoe, 2023-12-03:
Another issue I am having with this is if I try to put all of the animal crossing amiibo .bin in a single folder, the device just crashes right when I try to enter that folder on the device. I don’t know how to fix that due to the issue we were talking about. 

> Oh, I get your point, the fact is that the Pixl.js file system has limitation on its full file path for a file, like /amiibo/animalcrossing/amiibo1.bin. The total length of the file path string should not be exceed 64 characters, and file/folder name should not be exceed 48 characters. This limitation is introduced by SPIFFS file system.
> 
> ```
> #define VFS_MAX_NAME_LEN 48
> #define VFS_MAX_PATH_LEN 64
> #define VFS_MAX_META_LEN 128
> ```
> 
> Please shorten your folder name and file name, and keep the folder count minimum as you can.
> 
> Meanwhile, you can use iNFC update to latest firmware 2.8.3, it introduce a new app called Amiibo Database with almost all amiibo files built in, you can have a try.
> 
> Please let me know if you have more questions.