solosky/pixl.js #112: Allow updating the database separate from the firmware

DanTheMan827, 2023-08-31:
The amiibo database could be stored on the storage in a file, and when the web interface connects, it could automatically upload the newest database.

This JSON could be converted to a binary format to save on storage space.

https://[n3evin.github.io/AmiiboAPI/database/amiibo.json](https://n3evin.github.io/AmiiboAPI/database/amiibo.json)

solosky, 2023-09-01:
Actually, the database in Pixl.js is already customized for size and manually categorized to games according to common use scenarios.  It adds more information like item drop and also have translated to chinese and english. 

The amiidb_amibo.csv is built from https://www.amiiboapi.com/ 

It kinds of complicated if we support external database, but I will consider your suggestion in futhure release.

Culkerz, 2024-03-25:
Is there any way to change the sort method or would this be a new issue? I was happy to see that Sora was added, but really odd that he now shows up at the very bottom instead of alphabetically.