solosky/pixl.js #116: Crash after using amiibo db followed by using emulator menu

DanTheMan827, 2023-09-12:
Firmware v2.8.4

https://github.com/solosky/pixl.js/assets/790119/2a5b5885-c51b-4f65-912c-7a10c452723c

solosky, 2023-09-12:
Hi , I am also noticed this bug , it can be reproduced by my device.

This issue seems related to memory allocation issue, maybe some memory region is corrupted. It’s kind of hard to address this issue.

Thank you report the issue, I will continue working on this.

impeeza, 2023-10-20:
@solosky and @DanTheMan827 Sorry for beeing that guy, but on my device (allmiibo) with FW 2.8.5 this error still happen.

Even just now learning and playing with the "Amiibo Database" after a while on it when selecting the "Exit"  the device restarted also.

solosky, 2023-10-21:
I will let the issue opened, and I still cannot figure out how to fix the issue...

jumkey, 2023-10-21:
@solosky 我之前测试了下 把tlsf_free 换成普通的free就不会崩。好像是clear favlist的时候调用了free

solosky, 2023-11-18:
Hi everyone,  

I think I finally fixed this issue, would you please have a try the fix version: https://github.com/solosky/pixl.js/actions/runs/6911499606

impeeza, 2023-11-18:
wow will tried just now and worked great.

commit 2721e00b (2023-09-15): fix #116
commit 4ac96aa0 (2023-09-15): Merge pull request #117 from jumkey/hotfix/fix116

fix #116
closed 2023-11-19 by solosky