solosky/pixl.js #119: Backup Current Firmware / Files

impeeza, 2023-10-19:
Hello there, I just Acquire a pre-made kit named «allmiibo» as far I can see is highly based on the original Pixl.js hardware, as far I can see it have a "modified" 2.6.2 firmware and is on English language.

Actually the device shows the next on System Settings -> Version
```
Version [2.6.2]
BuildDate [2023-07-2
GitHash [0783e4d]
Branch [main]
Target [0]
Dirty [1]
```
I want to install newer pixl.js firmware on it but thinking what the firmware and hardware could be very different from the original design, I like to have a backup of the current status of the device in order to be able to return to a well know working state.

How can this be achieved?

Thanks a lot for all your help.

impeeza, 2023-10-19:
Thanks a lot for the quick response.

Perfect, the upgrade process seems straightforward, but I really like to have a copy of the actual firmware in case something goes wrong I will be able to go back to the actual state.

Regards.

impeeza, 2023-10-20:
Hello there, I just updated the firmware to 2.8.4, and works fine, testing all features and so far working well, The only issue is what now all description tags of all amiibos are on Chinese, even if the language settings are on English, on the original firmware all descriptions was on English.

Thanks a lot for your responses and help.

solosky, 2023-10-20:
Hi, the language of tag description should be English if you choose English in setting menu.
Would you please to post more information, like screenshot to let me have better understanding the issue?

impeeza, 2023-10-20:
Hello there, I think I get confused myself, lets try to unravel my mind :P 

My settings are like this:
[image]

so If I go Main Menu -> amiibo Emulator
[image]
I got into a "File Explorer", entering on the "amiibo" folder
[image]
and then into other folder:
[image]
I can select a amiibo bin file:
[image]
And for that amiibo the description tag is on English:
[image]

but if I go to amiiboLik folder and select one of the files on that folder
[image]
[image]
The details of that file are on Chinese:
[image]

Also if on the Main Menu I go to AmiiboLink
[image]

And then Select one of the files:
[image]

The Tag details also are on Chinese:
[image]

therefore, seems is a problem with the files on the AmiiboLink folder, the ones transferred by the AmiLoop app.

solosky, 2023-10-20:
Hi, interesting findings, I will try it later, because I don't have device on hand now. 

BTW, you can try the Amiibo Database app, it contains almost all released amiibo information, and generate amiibo data on the fly. Also, it has more detail description for Zelda game series amiibos.

impeeza, 2023-10-20:
Hello there, I just downloaded and install on my android phone the Umiibo app, and transfer some bin files on AmiiboLink mode and the descriptions are on chinesse :(

The Amiibo Database App is Umiibo?

solosky, 2023-10-20:
No, I mean the app in Pixl.js. 

[image]

solosky, 2023-10-20:
Hi, I confirmed it's a bug. 
The tag description did not honor the language settings.
I will release a new version later today to fix this issue.

Thank your report on this.

closed 2023-10-20 by solosky

solosky, 2023-10-20:
Hi, there, please update firmware to 2.8.5 to fix this name issue.

impeeza, 2023-10-20:
@solosky sorry for ping a closed issue, I still have the doubt about the " Amiibo Database app" you mentioned on a message above.  which one is that app?  thanks.

impeeza, 2023-10-20:
> No, I mean the app in Pixl.js.
> 
> [image]

Thanks, learing about that section, great addition to the device.

wouter-luimes, 2023-11-10:
> Thanks a lot for the quick response.
> 
> Perfect, the upgrade process seems straightforward, but I really like to have a copy of the actual firmware in case something goes wrong I will be able to go back to the actual state.
> 
> Regards.

Sorry to bother you, but were you able to obtain a copy of the firmware that originally came with the device in the end? And if so, would you be able to share it? I have this same device myself and also would like to have a back-up just in case.

impeeza, 2023-11-10:
> Sorry to bother you, but were you able to obtain a copy of the firmware that originally came with the device in the end? And if so, would you be able to share it? I have this same device myself and [...]

No, I was unable but never needed, the device works very well with 2.8.6 firmware and the improvement are welcomed.

commit 78ad0177 (2023-10-20): fixed #119