solosky/pixl.js #122: Menu Animation Will not Turn Off

BladeTechBlade, 2023-10-26:
on version 2.8.5 i like the new update but what is menu animation and how do i turn it off

BladeTechBlade, 2023-10-27:
Because it will not turn off it stays on.

solosky, 2023-10-28:
Sorry, tested via LCD and OLED version, but I cannot reproduce this issue, would you please provide detail step on how to reproduce this issue?

BladeTechBlade, 2023-10-28:
What I mean is the reason why I wanted to turn it off is because on version 2.7.0 it is faster with it on so scrolling through the amiibos faster But when doing it with the 2.8.5 it is running slower to scroll show yeah so that's why I need it off to see if it made it get faster otherwise I might have to switch between 2.7.0 and then to the latest But for safety I only switch between 2.7.0 and 2.8.4 till it has been resolved.
 I got some good ideas that you could have when you run the new version  Give us some options like to turn off the menu Thing but when I add my amiibos I don't like the fact that I have to I have to  Rename my own amiibos Can you do an app to rename the amiibos but with the 63 character limit for each of them please because rename in them is gonna take a while to do because I have the entire collection from which I own and IM using this device to store them on and on the device is so good I would Cause I've had to shorten the names just to put them on so then the scrolling does help a bit but not fully like when I go into was at 2.8.5 the database It is also slow when it is under the latest version to show me the amiibos items and what you get is a good that I want to use that Feature Or add my allmiibo zero with lcd screen is what I got.
If you need pictures sent please ask me but a video would be better it would take time to do it because I would have to but firmware back on to show you.

solosky, 2023-10-28:
You can turn the animation via "Menu Animation" menu in Settings if you feel the animation is annoy. 

Regarding the 63 name limitation, sorry for that, but it's a limitation from SPIFFS, you can try to name the folder as short as possible, so there may be small amont of amiibo names that requires shorten its file name. 
Another way is that using some file rename tools to rename your amiibo files. 

Amiibo database is another choose for amiibo file generation and emulation, you can have a try.

I want highlight that, the change between 2.8.5 and 2.8.4 is a minor bug fix change,  I dont think so it will impact your scrolling speed if you upgrade your device from 2.8.4 to 2.8.5.

x7l7j8cc, 2023-10-28:
@BladeTechBlade To be able again to Set the Menu Animation, just:
1. Connect via "Web Bluetooth Transmission" and Delete the **settings.dat** file in the **_External Drive_**.
2. Reboot. And you should have the setting set to OFF, and now you can turn on or off as you wish. 
Obviously, you need to set the language and backlight again.

impeeza, 2023-10-30:
Hello there, for me was the settings.bin file which have the value of 03 on the 0x0C offset, after delete the settings.bin file, and put the same settings like before I got these values:
[image]
the file at the left is the recreated settings, and the one on the right is the original one which doesn't allows to change the Menu Animation Setting.

the 0x28 value on 0x0B offset I don't know what means.

solosky, 2023-10-30:
The 03 on the 0x0C should be the bool anim_enabled, it should be 1 or 0, 3 is wrong value here(I don't know why 3 is here..).
0x28 is oled constract..
just for your reference, the settings.bin layout as bellow:
```
typedef struct {
    bool backlight; //deprecated, keep for capability issue
    uint8_t sleep_timeout_sec;
    bool dirty;
    bool auto_gen_amiibo;
    bool auto_gen_amiibolink;
    bool skip_driver_select;
    bool bat_mode;
    ble_amiibolink_ver_t amiibo_link_ver;
    bool hibernate_enabled;
    bool show_mem_usage;
    uint8_t lcd_backlight;
    uint8_t oled_contrast;
    bool anim_enabled;
    Language language;
    uint8_t amiidb_data_slot_num;
} settings_data_t;
```

Deleting the settings.bin to solve this issue should be fine, because the settings.bin will be saved when first time to go sleep.

BladeTechBlade, 2023-10-31:
Do you need what is showed in the text if you already deleted the settings.bin

solosky, 2023-10-31:
> Do you need what is showed in the text if you already deleted the settings.bin

sorry, I didn't get your question. Please explain a bit more?

solosky, 2023-10-31:
Not need, the device should be set to default value 0.

BladeTechBlade, 2023-11-09:
Thanks that has explained what I have asked for.

impeeza, 2023-11-09:
So, the issue can be closed now?

BladeTechBlade, 2023-11-10:
> So, the issue can be closed now?

Yep all done asking this questions.

BladeTechBlade, 2023-11-10:
⅚

BladeTechBlade, 2023-11-10:
Bye

impeeza, 2023-11-10:
he hehe, please close the issue :D

BladeTechBlade, 2023-11-10:
Thanks for everyone's help

scottjl, 2023-11-20:
2.8.7, still can't get the option to turn off. tried deleting settings.bin as recommended by @x7l7j8cc but doesn't seem to work. maybe add an option to reset all settings, or during firmware upgrade reset them? thanks!

solosky, 2023-11-20:
> 2.8.7, still can't get the option to turn off. tried deleting settings.bin as recommended by @x7l7j8cc but doesn't seem to work. maybe add an option to reset all settings, or during firmware upgrade reset them? thanks!

Delete the settings.bin has the same effect with reset all settings. 
I still cannot reproduce this issue. 
If you know how to reproduce this in detail, please let me know.

scottjl, 2023-11-20:
> Delete the settings.bin has the same effect with reset all settings.
> I still cannot reproduce this issue.
> If you know how to reproduce this in detail, please let me know.

deleting the file does nothing for me. actually i'm not even sure the file gets deleted because after i delete it and have the unit reset, i would expect the language to default back to chinese (?) but it doesn't and all my other settings are intact. it doesn't seem like the file is actually being deleted. i even tried to "format disk" and the settings persist.

not sure how to reproduce the issue, since it's been like this since i got my unit a week ago. i updated to 2.8.6 and 2.8.7 and it's still present.

i can also verify 0x0C is 03 on my unit. i've tried changing it and up-loading the altered file, but that doesn't seem to work either. after resetting and downloading the file, it's still 03.

if i had to guess, i'd say a handle to the file is still open from the OS and doesn't seem to let me overwrite or delete the original file. i need some way to force it closed to alter it outside of the settings menu. might just be easier to put in a check for a value of 0x0C of any value other than 0 or 1 and if found, force it to 1.

solosky, 2023-11-20:
can you try delete in the web portal ?

scottjl, 2023-11-20:
deleted via web portal (with Chrome). file looks deleted. went to settings and reboot. menus are still in English, still unable to toggle animation off.

solosky, 2023-11-20:
after delete from portal, try click refresh button to see is the file still exists or not

scottjl, 2023-11-20:
> after delete from portal, try click refresh button to see is the file still exists or not

it does not.

i delete file. refresh. then take unit out of BLE file transfer, go to settings, reboot.

scottjl, 2023-11-20:
for what it's worth, this is the unit i purchased. https://www.aliexpress.us/item/3256805821225710.html though mine looks slightly different and says 'allmiibo' on it. it loads pixl.js and runs it fine, except for this one issue. i have upgraded it using iNFC on iOS after downloading 2.8.6 LCD through the app and once downloading 2.8.7 LCD directly from this repo, saving it to iNFC and updating manually.

impeeza, 2023-11-20:
Ok, please do the Next, put the device in BLE File Transfer mode, and open https://pixl.amiibo.xyz/ then connect the device and share with us a screenshot of the root folder of flash storage.

then download settings.bin and share it with us.

select the settings.bin file with a tick on the left, then use the upper red button "DELETE".

restart the device from the settings  and repeat all again. thanks.

impeeza, 2023-11-20:
@solosky I don´t know if this could be a bug, if you change the settings.bin file on BLE File Transfer mode, when the device is reboot from settings menu, the just uploaded file is replaced by the one on memory.

impeeza, 2023-11-20:
I don´t know if put a code to reload the settings.bin file once you exit BLE File Transfer app, or recreate the default one if was deleted.

scottjl, 2023-11-20:
> so please do the next
> 
>     * Put the device in BLE File Transfer mode, and open https://pixl.amiibo.xyz/ then connect the device and share with us a screenshot of the root folder of flash storage.
> 
>     * Select the settings.bin file with a tick on the left, then use the upper red button "DELETE".
> 
>     * Without disconnect the device, press de DFU device to put it on DFU mode without exit the BLE File Tranfer mode. do not open the upgrade page, and wait until the device timeouts from DFU mode, a new settings.bin will be created.

This process worked for me. I couldn't get Chrome to connect, but I used iNFC on my iPhone. Followed the same steps. Once the unit timed out and rebooted it must have created a fresh settings.bin file, it booted up in Chinese, I was able to swap to English, and the Menu Animation option is now able to be toggled on and off.

Thanks for helping me get it fixed, there's definitely a bug somewhere.

solosky, 2023-11-21:
Hi @impeeza , it's a very interesting finding. I considered this as a bug, and I will try fix it in next version.

solosky, 2023-11-21:
Sorry, I try to reproduce this issue, but still cannot success, even I manually set the wrong settings value  03 on the 0x0C , it still can disable or enable menu annimation. 

I checked the code, it has settings_validate procedure call after setting read from settings.bin, if there is invalid value, it will reset to default. 

```
static void validate_settings() {
    if (m_settings_data.sleep_timeout_sec == 0 || m_settings_data.sleep_timeout_sec > 180) {
        m_settings_data.sleep_timeout_sec = 30;
    }

    if (m_settings_data.amiibo_link_ver != BLE_AMIIBOLINK_VER_V1 &&
        m_settings_data.amiibo_link_ver != BLE_AMIIBOLINK_VER_V2 && m_settings_data.amiibo_link_ver != BLE_AMILOOP) {
        m_settings_data.amiibo_link_ver = BLE_AMIIBOLINK_VER_V1;
    }

    BOOL_VALIDATE(m_settings_data.skip_driver_select, 0);
    BOOL_VALIDATE(m_settings_data.show_mem_usage, 0);
    BOOL_VALIDATE(m_settings_data.hibernate_enabled, 0);
    BOOL_VALIDATE(m_settings_data.bat_mode, 0);
    BOOL_VALIDATE(m_settings_data.skip_driver_select, 0);
    BOOL_VALIDATE(m_settings_data.auto_gen_amiibo, 0);
    BOOL_VALIDATE(m_settings_data.auto_gen_amiibolink, 0);
    BOOL_VALIDATE(m_settings_data.backlight, 0);
    INT8_VALIDATE(m_settings_data.lcd_backlight, 0, 100, 0);
    INT8_VALIDATE(m_settings_data.oled_contrast, 0, 100, 0);
    BOOL_VALIDATE(m_settings_data.anim_enabled, 0);
    BOOL_VALIDATE(m_settings_data.qrcode_enabled, 0);
    INT8_VALIDATE(m_settings_data.language, 0, LANGUAGE_COUNT - 1, 0);
    INT8_VALIDATE(m_settings_data.amiidb_data_slot_num, 1, 100, 20);
}
```

It may be a rare case for this, I will not continue address this issue, until more case reported.

closed 2023-11-21 by solosky

scottjl, 2023-11-21:
well if it happens to me again i know how to fix it. maybe add this as an item to a FAQ/Wiki in case someone searching for help doesn't find it here.

impeeza, 2023-11-22:
Hello there, I think I just find something weird:

The Aliexpress's Allmiibos devices comes with a custom 2.6.2 firmware, that firmware has been modified to some languages (Spanish, Italian, English and Chinese) they sell that variations, you can not change the language.

The settings.bin file included on that device is ONLY 15 bytes long:
[settings_bin.zip](https://github.com/solosky/pixl.js/files/13432609/settings_bin.zip)

As you can see the offset 0x07 is 0x03 on that file.

----

To reproduce the problem I just made the next:

Using BLE Transfer File Mode, deleted the current settings.bin file, then put the device on DFU mode.

After timeout the default settings are on the device (looking on the menus):

Sleep Timeout Timer	30 Seg.
Automatic Random	ON
Auto Random AmiiboLink	OFF
Auto Storage		OFF
LiPO Battery		OFF
AmiiboLink Mode		V1
Fast Resume		OFF
Show Memory Used	OFF
Backlight Brightness	OFF
OLED Contrast		
Menu Animation		OFF
Language		ZH_HANS(简体中文)
Amiibo Slot Num.	20

If after the timeout you immediately go to BLE File Transfer again, there is no settings.bin file on the root of the Flash Storage.

Changed the language to English, to test, and reboot the device,  a new 16 bytes settings.bin file is created.

Now on BLE File Transfer, Using red DELETE button erased the settings.bin and uploaded the "original" 15 bytes settings.bin, the again entered on DFU mode.

After the timeout on the device is on Chinese, then started BLE FT again and the 15 bytes settings.bin file is still on the root of flash storage.

exit the BLE FT mode and reboot the device, now the settings.bin file is 16 bytes long.

Now go to settings, and try to change the menu animation and stays OFF no matter how many times I press the middle button:

https://github.com/solosky/pixl.js/assets/41979604/4b9a28b9-5a69-41b1-87b3-2d0207917e6a

go to BLE F T and delete the settings.bin file, restarted the device and no avail still no able to change menu animation.

the only way to be able is going to BLE F T mode, delete the file and then going to DFU mode, after that again a default settings.bin file is created and now you can change the menu animation settings.

impeeza, 2023-11-22:
Extra test,

Downgraded the device to 2.6.2 firmware, deleted and restarted (battery) the settings.bin file, a new settings.bin file 11 bytes long is created:

[settingsDefault262.zip](https://github.com/solosky/pixl.js/files/13433280/settingsDefault262.zip)

then upgrade to 2.8.7 and after a new reboot, a settings.bin file with 16 bytes size is created
[settings.zip](https://github.com/solosky/pixl.js/files/13433284/settings.zip)

the correct 0x07 offset is set up.

Definitively the problem is with the settings.bin file included with the Alliexpress allmiibo device which have the 0x03 value on 0x07 offset.

But is weird what if you delete the settings.bin file on the BLE File Transfer mode is recreated exactly the same after reboot.