solosky/pixl.js #125: Spanish Translation

impeeza, 2023-10-30:
Hello there,

I am working on a «es_CO.c» containing the Spanish translation of the «en_US.c» file which could be used to extend the languages of the device.

The problem I am facing right now are the special characters, because on Spanish we use «á é í ó ú ñ Á É Í Ó Ú Ñ» but even if the file «es_CO.c» that characters are encoded on UTF-8 are not displayed by the device.

I am not an expert programmer so trying to understand the code is hard to me.

Can you please tell me which characters should be on the file to that symbols get displayed on the device.

I am attaching the file I have so far, I am using the code es_CO to indicate Spanish - Colombia, but is the equivalent of 419 (Spanish Latam) I don't know which is the correct code.

[es_CO.c.txt](https://github.com/solosky/pixl.js/files/13202255/es_CO.c.txt)

solosky, 2023-10-30:
Hi, just let you know, I have updated the spanish translation in branch *spanish*, but unfortunately the special charaters still cannot be rendered correctly, I will make further invertigation to find out a solution if I am available.

impeeza, 2023-10-30:
Hello, as far I could look into the code, the FW is using the fonts 
siji_t_6x10, wqy12_t_gb2312a & wqy12_t_gb2312a_lite

this are the codepages?
Siji:
[image]

WQY12 t gb2312a:
[image]

What I don't finish to understand is how you convert the .c file onto UTF-8 to that code page.

Regards.

solosky, 2023-11-05:
Hi, I finally resolved the issue and the Spanlish is supported by develop firmware version, you can download the latest auto build firmware [here](https://github.com/solosky/pixl.js/actions/runs/6760673380)  to have a try.

impeeza, 2023-11-05:
Seizing the opportunity, I used the webpage to upload the DFU update, and worked perefectly.

Tested the new menus and translation and I am very happy.

the only tweak will be the one expressed on: https://github.com/solosky/pixl.js/issues/124#issuecomment-1793834105

solosky, 2023-11-06:
Updated as your suggestion, you can download new firmware from [here](https://github.com/solosky/pixl.js/actions/runs/6772097763)

impeeza, 2023-11-06:
Perfect!

Thanks for all your work

closed 2023-11-06 by impeeza