solosky/pixl.js #127: Web pages translation

impeeza, 2023-10-30:
Hello there, working on the firmware Translation I take to oportunity and translated the files of pixl.amiibo.xyz page, I took the files from SourceCode:/gh-pages and modify them, I don't know if having different set of files for each language, or if the files could be changed to be multilingual.

Also I don't know if the edition goes directly on the files
- pixl.js-main\gh-pages\index.js
- pixl.js-main\gh-pages\manifest.js
- pixl.js-main\gh-pages\vendor.js

or the text should be edited on other set of files.

In order to edit that files I did use a javascript Beautifier to get a more editable code.

I am attaching the set of files on English and Spanish
[gh-pages_ES.zip](https://github.com/solosky/pixl.js/files/13210349/gh-pages_ES.zip)
[gh-pages_EN.zip](https://github.com/solosky/pixl.js/files/13210350/gh-pages_EN.zip)

Let me know if I can help more.

solosky, 2023-10-31:
The content in gh-pages is built from vue web project (web directory).
You should not update the content in gh-pages, you should edit the vue files in web directory instead.

The site is a vue project, currently this project is not support i18n translation, you may introduce [Vue-i18n](https://vue-i18n.intlify.dev/) to have better i18n translation support.

impeeza, 2023-10-31:
Will search and learn what VUE is it😸

impeeza, 2023-10-31:
Hello there, vue.js is giving me a hard time.  so far using the files on SourceCode:/web/ I extract the needed strings to internationalize the webpages, I am Attaching a Spreadsheet with the location and strings:
[Strings.ods](https://github.com/solosky/pixl.js/files/13221995/Strings.ods)

impeeza, 2023-11-01:
Hello everyone, Just was able to build the Web Page, so after some attempts I got two set of Web Pages one on English other on Spanish.

Also Found what the code on the file «App.vue» after line # 350 is lacking of the names for the «Accept» & «Cancel» buttons, so always the default Chinese names are used.

So I modified the file like this:
Orignal one:
```    on_upload_diag_close(done) {
      this.$confirm('确认关闭？关闭将清空上传记录。')
        .then(_ => {
          this.$refs.upload.clearFiles();
          this.reload_folder();
          done();
        })
        .catch(_ => { });

```

And add the lines:

```
    on_upload_diag_close(done) {
      this.$confirm('确认关闭？关闭将清空上传记录。')
        confirmButtonText: '确定',
        cancelButtonText: '取消',
      })
      .then(_ => {
        this.$refs.upload.clearFiles();
        this.reload_folder();
        done();
      })
      .catch(_ => { });
    },

```

and after modifying the file, created this two files:
[AppEN.vue.txt](https://github.com/solosky/pixl.js/files/13223479/AppEN.vue.txt)
[AppES.vue.txt](https://github.com/solosky/pixl.js/files/13223480/AppES.vue.txt)

But could be better to modify the file to use external defined variables, or having a combo or buttons to select the different languages.

Additional there are some text I couldn't translate because seems not to be defined on the app, and maybe using default values from the modules or somewhere else:

Creating a new folder failed:
[image]

Error renaming a object because new name too long:
[image]

When there are no data:
[image]

The last version of the strings changed are summarized on the file
[Strings.xlsx](https://github.com/solosky/pixl.js/files/13223490/Strings.xlsx)

impeeza, 2023-11-01:
Found the text about "No Data" on a empty table, looking on the page https://element.eleme.io/#/en-US/component/table realize what the property `empty-text` have the value, so needs to be added to the code.

The line 46 of the file `App.vue`, actually have:

```      <el-table ref="multipleTable" :data="tableData" tooltip-effect="dark" style="width: 100%" v-loading="table_loading"````

and if you add ` empty-text="No Data" ` to the header the text is customized.
```      <el-table ref="multipleTable" :data="tableData" empty-text="No Data" tooltip-effect="dark" style="width: 100%" v-loading="table_loading"````

Also I am trying to understand https://element.eleme.io/#/en-US/component/i18n about internationalization of `vue` projects.

solosky, 2023-11-02:
Hi, glad you have made such great progress. It looks promising and goes on the right way. 
I am waiting for you further progress update.

impeeza, 2023-11-02:
I am trying to understand how i18n works on vue & element but I am not really a javascript programmer and I am a little lost.

Which framework or editor are you using for the web project?

solosky, 2023-11-02:
definitely vscode.
I use vscode for web developing and firmware developing both.

impeeza, 2023-11-02:
Ok, Got more lost here, but will try to learn something.

impeeza, 2023-11-05:
Well, As my first venture on a "real" programming (I do normally only do VBScript and Powershell) I just made the Pull request https://github.com/solosky/pixl.js/pull/128 to add the changes on the Web page Pixl.js

Was a learning journey, a good one.

I tried to be respectful with the actual programming syntax and style.

The file `zh_TW.js` so far is a copy of `zh_Hans.j`s because my Chinese skills are null and ask you to made the needed changes to the file.

@solosky please review the proposal and let me know if the changes are suitable to final product.  I did take the freedom of made some little changes on the code no related to language and hope they aren't too intrusive.

I made some test to the page on with my device but some scenarios couldn't be tested.

Hope this PR will be useful on the future.

solosky, 2023-11-05:
The PR looks promising, it's a great enhancement for the web application.
I am working on your PR, and if no issue is found, I will deploy to online.

Again, thank for your great contribution on the Pixl.js project.

solosky, 2023-11-05:
Hi sir, 

Thank your great contribution, the i18n support of web site is deployed to live now,  you can preview the site with links bellow:

* https://pixl.amiibo.xyz/
* https://solosky.github.io/pixl.js/

I also have do some update and fix some bugs to have better user experience for this great enhancement:

  1. Language drop menu always show same content
  2. Save language selection to cookies as preference
  3. zh_TW transalation via ChatGPT!
  4. update npm dependecies in package.json

Please let me know if you have futher comments.

Once again, thanks for your contribution to the Pixl.js project!

impeeza, 2023-11-05:
Wow that was quick, excellent catching on that label what eluded my eyesight 😄 
The cookies is a very nice addition, I was afraid of use them.  even the cookies will easy the detection on the connected_version button.

Thanks a lot for all your work.  

Regards From Colombia.

impeeza, 2023-11-05:
Sorry, for reopen, I just spotted a strange bug, let walk trought the steps to reproduce it:

Started with a "incognito" browser tab and open https://pixl.amiibo.xyz/ or https://solosky.github.io/pixl.js/

[image]

as you can see the default language is zh_CN, cool, but the "Connect" button, is on English (my browser language), so if you select using the dropdown a different language, everything is ok:
[image]
[image]
but if you press ctrl+F5 to refresh the page, again the connect button get in to the browser language, by example on a English browser and last selected language: "Spanish":
[image]
or a Spanish browser and last selected language: "English:
[image]

Again Thanks a lot for your great work, Big Left Hand Shake from Colombia.

impeeza, 2023-11-05:
Thinks the bug was introduced by me on the web\src\i18n\index.js with the code managing the logic of selection of the default language, at first it checks and if not default language uses browser one, think the cookies should be tested there also. maybe.

solosky, 2023-11-05:
The issue is resolved and deployed to live.

impeeza, 2023-11-05:
Thanks a Lot, works like a charm 🐱

closed 2023-11-05 by impeeza