solosky/pixl.js #138: Need help using my Allmiibo Zero Device

BladeTechBlade, 2023-11-12:
This is my first amiibo emulator device and some Features I do not understand because their is not guide telling me how to use it.

solosky, 2023-11-12:
Hi, this is a opensource project, sorry the document is still not rich for user. 

I think you may have purchased Allmiibo (I dont know where do you buy from, but I dont sale any Pixl.js like device) from dealers, you could ask detail user manual for help from your dealers, because you already paid for your device. 

I would very happy if anybody could contribute to the the documentation.

impeeza, 2023-11-12:
I was thinking about state to documenting the basics, the next week will start

impeeza, 2023-11-12:
Even I was about to ask about the animation player and if you mind if I translate some .md files

BladeTechBlade, 2023-11-12:
> Even I was about to ask about the animation player and if you mind if I translate some .md files

i do not have .md files on my allmiibo zero device.

impeeza, 2023-11-13:
> > Even I was about to ask about the animation player and if you mind if I translate some .md files
> 
> i do not have .md files on my allmiibo zero device.

do not worry, is more a question to developers,  I you need help on the use of your device, you can:

* Read the actual documentation on the First page of GitHub, you can use google translate to help you if you don't read Chinese.
* Wait until the documentation project is more mature.
* Go to the QQ group, or a site like GBATemp asking for help.

impeeza, 2023-11-15:
I just started a «user manual» fort the Pixl.js device and firmware, I found some items for which I need some help and explanation, how can I contact you in a more fluid way?  Regards

solosky, 2023-11-15:
Hi, you may join the Pixl.js discord to have live chat with me. 

https://discord.gg/t8naCqy4