solosky/pixl.js #145: Any way to fix after flashing wrong firmware?

toaks2804, 2023-11-19:
Hello. I accidentally flashed the OLED firmware to the LED version. The screen lights up but there is no menu visible. Any way to save this?

impeeza, 2023-11-19:
you could try to move (blind) by the menus and put the device on the DFU mode, or use the wired mode to reprogram the device.

From the Main Menu you can put the device in DFU mode pressing:

_press any button to wake up the device, then press:_

- LEFT
- MIDDLE
- LEFT X 3
- MIDDLE

(Also to be sure you are on the MAIN MENU, you can remove and put the battery, then start the process above)

Open the page https://thegecko.github.io/web-bluetooth-dfu/examples/web.html load the `pixjs_ota_vXXX.zip` file for the LCD and open it, then press the `SELECT DEVICE` button and if your device is on DFU mode you should see a device called `pixl dfu` connect to it and the process should be finish well.

Also, from main menu you can press:

- LEFT x 2
- MIDDLE
then you are en BLE uplad mode then in https://pixl.amiibo.xyz/ page, you can connect and use the button `DFU` to put the device in DFU mode and programming again.

**For wire programming:**
This method requires the additional purchase of a CMASS-DAP compatible downloader. We recommend pwlink2 lite for 9.9 yuan.
Purchase address: https://item.taobao.com/item.htm?spm=a1z09.2.0.0.4b942e8deXyaQO&id=675067753017&_u=d2p75qfn774a

- Download the latest version of the firmware zip package and use JLink or DAPLink to program pixjs_all.hex.
- After connecting the cable, double-click `fw_update.bat` to program the firmware.
- After programming is completed, subsequent firmware updates can be updated via OTA.
- You can also use OpenOCD to program.
```
openocd -f interface/cmsis-dap.cfg -c "transport select swd" -f target/nrf52.cfg -d2 -c init -c "reset init" -c halt -c "nrf5 mass_erase" -c "program pixjs_all. hex verify" -c "program nrf52832_xxaa.hex verify" -c exit
```

toaks2804, 2023-11-19:
Thank you! That fixed it!!

closed 2023-11-19 by toaks2804

solosky, 2023-11-19:
@impeeza That's very detailed instruction, I will put your answer to docs,  thanks.

MayanKoyote, 2024-03-28:
> _press any button to wake up the device, then press:_
> 
>     * LEFT
> 
>     * MIDDLE
> 
>     * LEFT X 3
> 
>     * MIDDLE

looks like in case of modern FW 3rd step will be 'LEFT X 4' instead of 'LEFT X 3'

impeeza, 2024-03-29:
Yeah on newer firmwares that has changed a little bit.

R9anSmart, 2025-02-01:
Hi all,
I have also managed to make the classic mistake of flashing OLED to LED.

I have managed to flash the LED firmware which now gets the backlight working again but now I have nothing at all showing on screen. I am unable to remove the battery as it is soldered to the board so I will have to run the battery down to get to a known position in the menus.

What I really need to know is - why did the OLED or LED firmware work and how do I get back to a working state?

Thanks in advance

MayanKoyote, 2025-02-01:
[image]

impeeza, 2025-02-16:
> Hi all, I have also managed to make the classic mistake of flashing OLED to LED.
>
> I have managed to flash the LED firmware which now gets the backlight working again but now I have nothing at all [...]

Please read the description of the release for latest firmware:

https://github.com/solosky/pixl.js/releases/tag/2.12.0

> For LCD versions, if the screen appears too faint after updating, you can reset the settings by following these button presses after the device enters standby mode: Middle, Left, Middle, Right (x3), Middle, Right (xN) until the screen becomes clear. Set the contrast to 80.

R9anSmart, 2025-02-16:
I think what I did is flash a older version of the firmware and the flashed the latest. For some reason that worked. Maybe reset contract controls doing that somehow