solosky/pixl.js #148: font_data_gen.sh error

impeeza, 2023-11-22:
Hello @solosky, on the file https://github.com/solosky/pixl.js/blob/develop/fw/docs/translation.md we are instructed to run the file `fw/scripts/font_data_gen.sh`

That file assume you are running from the folder `fw/` if you run the file from `fw/scripts/` the script fails, because it try to modify files on `application/src/`.

The script must be changed to look for files on `../application/src/` or modify the `translation.md` to change the text in order to instruct to run the `fw/scripts/font_data_gen.sh` file from `fw/` folder.

solosky, 2023-11-22:
Yes, it's a known issue, I will update the transalation.md later today.
Also, I will update the script to make sure it could be run in any work directory.

impeeza, 2023-11-22:
Excellent, Thanks for take the time to read this. so if you are working on that, I will close this issue, let me to know if I can help.

Regards

closed 2023-11-22 by impeeza

solosky, 2023-11-22:
Hi there, just wondering have you update the translation successfully?

impeeza, 2023-11-22:
I think I do,  running the `fw/scripts/font_data_gen.sh` script from the `fw/` folder I got new set of files including `application/src/mui/u8g2_font_wqy12_t_gb2312a.c`

solosky, 2023-11-22:
Good, looks great!
The instruction works as expected!

impeeza, 2023-11-22:
yeah, the only addition I will make is, that instructions only works under Linux or MSYS 😆

solosky, 2023-11-22:
correct, that's the limitation for current script, I think it could be rewrite into pure python script so it can work in all platforms. Let's put it into my roadmap.

impeeza, 2023-11-22:
My respect!!  I am light-years from be able to create that type of code. :D

solosky, 2023-11-22:
I have updated the script font_data_gen.sh to allow run the script in any work directory.

See https://github.com/solosky/pixl.js/commit/4bf362ab34718e0f7b051a36213a6154a6fa6f29