solosky/pixl.js #161: Reset Default Setting  bug

impeeza, 2023-12-01:
The new option in settings "Reset Default Setting" It's a great addition. I think I found a bug,

If you go to settings ->  Reset default settings and then you select Back to Main Menu, The device seems to reboot, but the current language of menus and most of the previous settings are still in place, however if you go to settings app again, the default values are shown on each item, only after a system reboot the default settings enter in place

Video and first report on the Discord channel>
https://discord.com/channels/1146978069609975890/1146978069609975894/1180058782047408158

solosky, 2023-12-03:
fixed in https://github.com/solosky/pixl.js/commit/9136d02c8f26e62698c1f923f3818814dde21a9e

impeeza, 2023-12-03:
Wow!, thanks.

commit 9136d02c (2023-12-02): fixed #161: apply setting after reset setting
closed 2023-12-03 by impeeza