solosky/pixl.js #163: Delete confirmation dialog

impeeza, 2023-12-03:
Trying to delete a BIN file on amiibo Emulator APP from file details sub-menu the confirmation dialog do not show the complete confirmation text, miss the first character.

And if the file name is longer than 15 characters long, is not completely shown.

May be auto warp the filename to second line or third line.

But if you try to delete from amiibo details sub-menu, the first character of the confirmation is shown. but also need to warp the filename.

impeeza, 2023-12-03:
File properties submenu
[image]

amiibo details submenu (for the same BIN file)
[image]

solosky, 2023-12-03:
Fixed in develop branch.

commit b8794a5a (2023-12-10): fixed #163: auto wrap text for dialog message