solosky/pixl.js #164: Wrong alignment on some options of menu items

impeeza, 2023-12-03:
For the sub-menu: Amiibo Database -> Tag Details, the options
* Auto Rand.
* Show QR
The current option shown between [ON/OFF] is not right aligned
[image]

For the sub-menu: amiiboLink -> amiibo Details, the options
* Mode
* Random auto.
* Compati. Mode
 
The current setting [ON/OFF] is not right aligned
[image]

solosky, 2023-12-03:
Yeah, you are right, I have introduce a new format for list view, but I haven't updated other senarios, I think I will update them in furture commits.

solosky, 2023-12-03:
Fixed in develop branch.

impeeza, 2023-12-04:
Thank you so much.

commit 1265f232 (2023-12-03): fixed #164: right align for amiidb menu
commit e0ca8cdd (2023-12-03): fixed #164: align right for amiibolink menu
commit 80437b45 (2023-12-10): fixed #164: right align text for version menu
closed 2023-12-04 by impeeza