solosky/pixl.js #173: Web Page deployment

impeeza, 2023-12-06:
Hello there.

Yesterday I noted if I deploy the web page generated since version 2.8.7 on a subfolder of the site, the site doesn't load, on the navigator console you get the errors:
[image]

Note the error is about to load some files on https://impeeza.github.io/, but my deployment is on https://impeeza.github.io/pixl.js/
```
GET https://impeeza.github.io/manifest.js?24ca9d7d1b6f5b9ab586 net::ERR_ABORTED 404 (Not Found)
pixl.js/:14
GET https://impeeza.github.io/vendor.js?dee6c97d367fe20331d2 net::ERR_ABORTED 404 (Not Found)
pixl.js/:14
GET https://impeeza.github.io/index.js?7861d874e3386936ff77 net::ERR_ABORTED 404 (Not Found)
favicon.ico:1
GET https://impeeza.github.io/favicon.ico 404 (Not Found)
```

The same happens on my PC, I have setup a Apache server, and if I setup https://192.168.100.8/pixl.js the site doesn't load, but if I setup all on the root site, works fine:
[image]

[image]

Even if I leave the files on the root site, and the subfolder, opening the subfolder load the site on the root:
[image]

There is anything can be done?

solosky, 2023-12-06:
You may change the `const publicPath = '/'` to `const publicPath = ''` in the screenshot bellow and then run `npm run build`  to build a new version. 

[image]

impeeza, 2023-12-06:
Will try it tomorrow, thanks a lot.

Will investigate how to put there a variable pointing to initial URL. regards.

solosky, 2024-01-17:
since there is no activity for a long time, I am closing this issue as resolved.

impeeza, 2024-01-17:
Thanks I was unable to make the new router to work so no news by now.

solosky, 2024-01-17:
Have you tried my solution?
It tested working on my laptop.

impeeza, 2024-01-17:
Didn't work on my setup, but also nothing worked so far, nor this project nor other projects so I messed up something on my setup, I am hopping next week deploy again all.

solosky, 2024-01-17:
I have re-opened this issue. 
Please let me know if there is any update on this.

impeeza, 2024-01-28:
Sorry I still no have PC the illness has having me a little slow.

solosky, 2024-01-28:
Take care yourself.

closed 2024-06-13 by impeeza