solosky/pixl.js #177: Menu icons

impeeza, 2023-12-09:
Hello there, I am wondering.  reading some code do reference to icons, by example the function add menu item (mui_list_view_add_item) one of the parameters is the icon.

Then I found the `mui_icons.h` which reference some of the used icons on other files. but the set is reduced, on some `.c` files the icon is directly the Hex value.

I wonder, How that Hex values are referenced to the font file?

Because if I do open the `.bdf` files using FontForge and do not find most of the icons.

Thanks a lot for your help.

solosky, 2023-12-10:
Hi, you can view the icon of hex code via https://github.com/solosky/pixl.js/blob/main/fw/docs/u8g2_font_siji_t_6x10.png

impeeza, 2023-12-10:
Thank you so much, I was blind, how I wasn't able to see that file :P

closed 2023-12-10 by impeeza