solosky/pixl.js #188: 可以支持mfkey32v2吗？

yunsur, 2024-01-08:
相关项目
https://github.com/equipter/mfkey32v2

https://github.com/solosky/ChameleonUltra/tree/938f2b312c92153fd7f1f598a162af9c77ee615f/software/src

solosky, 2024-01-09:
最近忙于做新的硬件版本，固件更新暂停了一段时间，这个功能我这边计划会做，可以点个watch关注下。

yunsur, 2024-01-09:
已经watch关注，感谢回复。

fishybow, 2024-02-12:
> 最近忙于做新的硬件版本，固件更新暂停了一段时间，这个功能我这边计划会做，可以点个watch关注下。

@solosky Did you finish making the new hardware? Is there anywhere I can buy it?

Jacky10X, 2024-04-27:
> 最近忙于做新的硬件版本，固件更新暂停了一段时间，这个功能我这边计划会做，可以点个watch关注下。

你好 請問關於新硬件有沒有新消息? 還是已經發售?

Jacky10X, 2024-04-29:
> 最近忙于做新的硬件版本，固件更新暂停了一段时间，这个功能我这边计划会做，可以点个watch关注下。
@solosky 你好 請問關於新硬件有沒有新消息? 還是已經推出?

solosky, 2024-04-29:
> > 最近忙于做新的硬件版本，固件更新暂停了一段时间，这个功能我这边计划会做，可以点个watch关注下。
> 
> @solosky 你好 請問關於新硬件有沒有新消息? 還是已經推出?

还没正式发布，硬件部分已经完成了，目前还在编写新硬件的固件。具体发布时间不确定，主要是因为我个人空余时间有限。

Jacky10X, 2024-05-07:
> > > 最近忙于做新的硬件版本，固件更新暂停了一段时间，这个功能我这边计划会做，可以点个watch关注下。
> > 
> > 
> > @solosky 你好 請問關於新硬件有沒有新消息? 還是已經推出?
> 
> 还没正式发布，硬件部分已经完成了，目前还在编写新硬件的固件。具体发布时间不确定，主要是因为我个人空余时间有限。

好的 感謝回答 還有一個問題 就是新硬件會只有一個版本還是有分幾個版本? 例如好像現在是有分LCD版跟OLED版

Jacky10X, 2024-05-08:
> > > 最近忙于做新的硬件版本，固件更新暂停了一段时间，这个功能我这边计划会做，可以点个watch关注下。
> > 
> > 
> > @solosky 你好 請問關於新硬件有沒有新消息? 還是已經推出?
> 
> 还没正式发布，硬件部分已经完成了，目前还在编写新硬件的固件。具体发布时间不确定，主要是因为我个人空余时间有限。

@solosky 好的 感謝回答 還有一個問題 就是新硬件會只有一個版本還是有幾個版本 例如會有LCD版跟OLED版?

Jacky10X, 2024-05-08:
@solosky 好的 感謝回答 還有一個問題 就是新硬件會只有一個版本還是有幾個版本 例如會有LCD版跟OLED版?

solosky, 2024-05-08:
只有OLED版本，新版本图片可以参考 #213 第一张图片。