solosky/pixl.js #194: Optimize memory consumption of language supports

fishybow, 2024-01-11:
While exploring locally, I discovered that adding more languages (and the corresponding translations) would cause the Chameleon app to crash, especially when the tag type is set to Mifare 4k.

I cannot be certain of the cause, but my suspicion is that the crash is caused due to out of memory. As it stands right now, all translation strings are loaded into memory by default because they are statistically compiled into the executable.

Action Items

- We should stop adding more languages until this investigation concludes.
- We should consider removing some of the less popular languages that have already been added.
- We should search into a better localization approach to reduce memory footprint. The ideal case would be only the active language is loaded into memory. The translation of other languages should exist in the form of files (either in the internal or the external storage). Those files are loaded on-demand as the user switch to that language.

solosky, 2024-01-11:
HI @fishybow, you are correct, the language const should be put into text section but somehow the linker put them into text section, see screen bellow.
[image]

solosky, 2024-01-11:
Thank your information, I have fixed this issue in #[7355f93](https://github.com/solosky/pixl.js/commit/7355f9301426221f87643cdf1d33b26f4229a998)
All language string now puts into the text section, no memory is consumed. 
It's a simple and tidy solutions for this issue. 

[image]

solosky, 2024-01-11:
Hi @fishybow , regarding the crash when enter Chameleon app, I think I have fixed that issue in commit https://github.com/solosky/pixl.js/commit/01d2c975c7f6156e143491a0fa93a0a14745c3b6, can you have a try the latest develop firmware and confirm the issue still did exist or not? (You can download develop firmware in Github Actions tab).

The issue seems not related with the language files. Anyway, the current language translations still can compile to final firmware, because a language file only occupied 0x294 bytes in flash, that's still acceptable.

fishybow, 2024-01-11:
Great thanks for the reply. Marking this as resolved.

commit 7355f930 (2024-01-11): fixed #194: put language string into text area to reduce mem footprint
closed 2024-01-11 by fishybow