solosky/pixl.js #197: Bug after updating to 2.9.3

Brickleberry19, 2024-01-13:
After I updated to version 2.9.3 it looks like this for me 
https://github.com/solosky/pixl.js/assets/115247704/858ca35b-c4f1-40b0-850b-3f76c2afe586

(OLED version Updated With nRF Connect on Android)

solosky, 2024-01-13:
May I know where did you download the firmware ? do you download the firmware from github release page?
I tested the release version from github release page,  no issue was found.

Brickleberry19, 2024-01-13:
Yes i Download it from your GitHub Release page

closed 2024-01-13 by solosky

solosky, 2024-01-13:
Hi @Brickleberry19, sorry, I think I have fixed this issue in 2.9.4, please upgrade to 2.9.4  https://github.com/solosky/pixl.js/releases/tag/2.9.4 

Thanks for your reporting this issue.

Brickleberry19, 2024-01-13:
Thanks for the quick fix, it works

solosky, 2024-01-13:
you are welcome

commit 168c711b (2024-01-13): fixed #197: wrong layout if animiation is disabled
PR #198 (2024-01-13): 2.9.4
fixed #197: wrong layout if animiation is disabled