solosky/pixl.js #199: Can't put whole animal crossing set into folder

Billybobertjoe, 2024-01-13:
This is a continuation of this issue: [https://github.com/solosky/pixl.js/issues/111]

If I try putting the whole animal crossing series of amiibo.bin files, I can't open the folder on the pixl or it makes it restart(crash?). I can't sort them into folders decently due to this issue: [https://github.com/solosky/pixl.js/issues/111]

I'm not sure what to do because I don't want to have it too unorganized

solosky, 2024-01-13:
Hi, the best usage for AC is Amiibo Database App, I have already categorized the AC figures and have a neat search function to find specified figure. 

If you would like to manage your own amiibo files using Amiibo Emulator App, please note the restriction for the file system, and dont put over 100 files in a single folder, it will make the device crash and restart.

solosky, 2024-01-13:
You can view User Manual of Amiibo Database App via https://github.com/solosky/pixl.js/blob/main/docs/en/04-Using-Firmware.md#amiibo-database

closed 2024-02-04 by solosky

impeeza, 2024-02-04:
Even 95 files will make the device crash :D 

Create subfolders