solosky/pixl.js #203: Open up more RFID / NFC slots

fishybow, 2024-01-16:
The card emulator has 8 slots max right now. Are there any concerns to open that up to more slots?

solosky, 2024-01-17:
Yes, if we change more card slots, slotConfig may consume to much mem and slower initiation for emulation.

But we may increase to a larger number like 64 slots as maximum slots allowed.  User can choose any number bellow 64 to be his own slot count.

fishybow, 2024-01-17:
That makes sense. We can keep 2 slots open by default but allow users to turn on a maximum of 64.

fishybow, 2024-01-31:
I looked at the code, it seems that the count 8 is defined by within a dependency library. There is no straight forward way to modify that. I'm closing this issue for now.

closed 2024-01-31 by fishybow