solosky/pixl.js #207: Reset device confirmation

PonyOny, 2024-01-20:
I purchased a "Allmiibo" some days ago, I have been updating it to the latest version several times since purchased, I don't know if this is something that only happens to me on my device but the "Top Button" the one that lets you navigate Right/Left/Confirm lots of times when pressing it it goes Left/Right then confirms, I have to be very steddy to press the confirm button without triggering the directions as well, what's the main problem with this, every time I enter the Settings menu and have to go back using the bottom menu item "back to Home" it goes up and resets the device to default state... this is really annoying as I have to go again and configure everything I changed and then cross my fingers this does not happen again when trying to go back to home screen, on other menus it's also annoying but not so much as when it restes de device, my point is, would it be possible to add a second step when you select the reset menu so you have a final confirmation, it would probably happen the same.. pressing Cancel then switching to Confirm.. but maybe you would have a second opportunity... or maybe there is another alternative to navigate back.. some sort of long confirm button press or something similar, I hope I've explained myself.

Thanks.

solosky, 2024-01-20:
Hi there, I think you are the first one to report this issue, I don't have any issue when go back to home when in settings APP. 
Anyway, your suggestion seems reasonable,  and I will add a confirmation message box before reset to default settings.

impeeza, 2024-01-20:
Cool. 

And the order of cancel accept commit to always accept to the dame side😸

solosky, 2024-01-23:
Sorry, I don't your point, the button should be layout with LEFT, CENTER, RIGHT
you can call ```mui_msg_box_set_btn_focus(app->p_msg_box, 0|1|2)``` set default focus icon.

solosky, 2024-01-23:
That's exactly order what I have done here. No CENTER button is required here. 
```
mui_msg_box_set_btn_text(app->p_msg_box, _T(CONFIRM), NULL, _T(CANCEL));
mui_msg_box_set_btn_focus(app->p_msg_box, 2);
```

solosky, 2024-01-23:
That's a good question here, my short answer is that the default action should carefully select by its senario.
For example, create new folder, the default button should be confirm, but for delete action, the default action should be cancel.

closed 2024-01-28 by solosky

solosky, 2024-01-29:
Hi @PonyOny , I don't have such issue and no one else report this issue too, so I consider your wheel switch may be broken, you may need to replace one or to request your seller to replace a new one.

GlebDGT11, 2024-08-27:
Hi, @solosky , I have recently catch problem for my allmiibo, that I bought on 2nd of July and getting it in 29th of July. So, the problem: When I wanted to download latest firmware to 2.11.0 from 2.10.2, firstly screen looking like it's getting hammered, but not, I quickly put it into DFU Updater to install that firmware again, and while updating is getting finished, it just stops at 240k firmware bytes and started bootlooping in the DFU Updater mode, and it won't able to return to Main Menu

Uploading VID_2024_08_27_16_45_14_1262661556606660999115.mp4…

commit 7f7edee6 (2024-01-22): fixed #207, show confirm box when reset settings