solosky/pixl.js #213: Pixl.js buyer guide

solosky, 2024-02-04:
Hi everyone, 

Thank for your starring this project! 

As you may know, I don't make and sell Pixl.js device, but I hope everyone that like this project is able to get a Pixl.js compatible device in a affordable price, so if you already buy a Pixl.js like device, please kindly post your buy link here to make it easy to choose what you would like to get one. 

Thank your contribution on this. 

## here is an example:

* Buy Link: [Taobao](https://item.taobao.com/xxxx) 
* Price: XXX CNY

[image]

impeeza, 2024-02-04:
Buy Link:  https://shop.mtoolstec.com/product/pixl-with-oled-for-amiibo
Price (2024_02_04): 29.99 USD
Details: OLED, USB-C for charging and for Firmware Update via SDCLK, very small, White or Black

> [image]

LEO318x, 2024-02-10:
Buy Link: [https://item.taobao.com/item.htm?id=732982684854](https://item.taobao.com/item.htm?id=732982684854)
Price (10-02-2024): ￥150.00 ≈ $22.26 
Details: OLED, USB-C, Battery 180mAh

[image]

[image]

Schokobecher, 2024-04-03:
Buy Link: [AliExpress](https://www.aliexpress.com/item/1005006420542854.html)
Price (21-03-2024): $12.61
Details: LCD, USB-C, 180mAh LiPO Battery
[image]
[image]
[image]

Flashed fine via nRF DFU (iOS) and DFU via the Web Interface

fanoush, 2024-06-09:
After getting one for $14, aliexpress started to offer me this amiibo pixl device in "3 from $xx" ChoiceDay deals when I search for "pixl" so I got few more :-)
[image]
so try to search for "pixl" when thins choice day deal is active, maybe you'll get the offer too but it is random, today I don't get it

I also get regularly offer for the keychain one for $2.69, which is not bad for nrf52832 device with NFC too
here it is today
[image]

skylandersNFC, 2024-09-17:
Let me give some links as well :

https://www.aliexpress.com/item/1005006994020829.html

https://www.aliexpress.com/item/1005006260883608.html

https://www.aliexpress.com/item/1005006732518163.html

https://www.aliexpress.com/item/1005006214951877.html

They are all around **17 bucks**.

Part of the **[Skylanders on Allmiibo](https://skylandersnfc.github.io/Docs/Skylanders_on_Allmiibo/)** guide.

skylandersNFC, 2024-11-22:
[image]
https://www.amazon.com/Flashiibo-Emulator-Expandable-Unlimited-Rewrite/dp/B0CQWRPJL2

grimson73, 2025-06-14:
https://nl.aliexpress.com/item/1005008726926205.html
- pixl.js firmware v2.12.0 already installed, guess a recent hardware batch made.
- Key_Retail.bin preinstalled
- Internal battery
- Delivery (the Netherlands) was within a week and the in transit delivery status communication excellent.

[image]
*image distortion due to screenrefresh / camerashutter.

[image]
Protective plastic

[image]
Real tiny but manageable.

As I just received this device it seems well working and would recommend for purchasing. Will see about durability now.