solosky/pixl.js #222: The amiibo emulator, how to add information to your own amiibos?

Vorffeed, 2024-03-23:
Any way to enter the amiibo description information as it appears in the database, but in the amiibo emulator, where you put your own amiibo backub so it doesn't come up underneath just what that amiibo's name is?

impeeza, 2024-03-23:
You can edit the source and build for your self

https://github.com/solosky/pixl.js/tree/main/fw/data

Vorffeed, 2024-03-23:
> You can edit the source and build for your self
> 
> https://github.com/solosky/pixl.js/tree/main/fw/data
But I don't see anything there that is for the amiibo emulator, I only see for the database.

impeeza, 2024-03-23:
The emulator gets all data from the bin file and the name is read from database if ID of the bin file is on the database.

Vorffeed, 2024-03-23:
Well, it doesn't do it to me as you can see ><
Database
[image]
vs Amiibo Emulator
[image]

solosky, 2024-03-24:
Hi the note in Amiibo Emulator does not display note in Amiibo database,  but you can add your own note by web UI or iNFC APP.

impeeza, 2024-03-24:
[image]
[image]

[image]

Vorffeed, 2024-03-25:
ok, thanks.

closed 2024-07-29 by solosky