solosky/pixl.js #223: Is possible to implement fimrware update via NFC cardreaders via PC?

Wohlstand, 2024-03-24:
Hello!

I do have such card reader that gets plugged via USB:
[image]
[image]

It works on Linux through the `pcsc` library (an output of `pcsc_scan` tool is here):
[image]

And I got a question: I do see already the use of iNFC thing on mobile devices, but what about PC support with the NFC devices?

impeeza, 2024-03-24:
May be possible, but practical I don't know,  the average firmware is 1.2 MB  and NFC normally transmit 1 Kb/s and you need that space free on the internal storage which is normally 1.8 MB so who knows!

solosky, 2024-03-25:
Possible, but not recomended, because comparing NFC, the BLE is most common hardware support by many device and nrf52 sdk and nrf toolset has built-in support for DFU by BLE.

Wohlstand, 2024-03-26:
Oh, about the NFC case I probably confused while I attempted to understand the manual 👀
I do have BLE adapter on my PC, and device gets clearly detected by Linux Mint's Bluetooth manager, however, to submit data need to have a program that supports custom protocols. I tried to run the nFR Connect thing, but it just can't detect anything (need to install dependencies to get it work properly).

impeeza, 2024-03-26:
You can use chrome on the Linux pc to upload firmware using the webpage

Wohlstand, 2024-03-26:
I already tried Chromium and M$ Edge, and it is unable to find anything, even at the specification of Web Bluetooth is noted that this API is experimental and is not yet properly standardised and desktop browsers doesn't implement it well. Maybe I'll try to make a kind of console tool that will use the raw Bluetooth directly? Just to upload/get files easily and send firmware updates vi DFU. Actually I wanted to build from the source code the modified version of firmware and try it out, but right now I waiting when a compatible programmator will arrive me. I wanted to try with Bluetooth way but I know the risk to brick the device without having a programmator.

Wohlstand, 2024-03-26:
Actually, using experimental APIs is 峨峨, using Web APIs that depends on minor set of browsers is 峨峨, and using Web that works on remote server for me is 峨峨 (I prefer standalone tools that will 100% work even Web server will be gone forever).

Wohlstand, 2024-03-26:
Anyway, how much is possible to turn the existing Web application into Electron-based app?

impeeza, 2024-03-27:
The website is based on VUE so may be because you also can create Electron apps using VUE.

Wohlstand, 2024-03-27:
> The website is based on VUE so may be because you also can create Electron apps using VUE.

I trying that to do now, first I need to tweak some files to make them not fail... Actually, Electron has working Web Bluetooth thing, so, it's possible that it WILL work.

Wohlstand, 2024-03-27:
Actually, for Electron case needs some small refactoring :thinking: : https://www.electronjs.org/docs/latest/tutorial/devices

Schokobecher, 2024-04-04:
I don't see the point in turning the current web app into an Electron app, adding more Chromium doesn't really make sense seeing how bare-bones the app is.

The BLE portion works fine, as well as the [DFU](https://thegecko.github.io/web-bluetooth-dfu/examples/web.html) app in Vivaldi on macOS, so I would assume the Linux Chromium should also work

Wohlstand, 2024-04-04:
The latest Chromium just fails:
[image]

Google Chrome fails too:
[image]

Yandex Browser too:
[image]

Actually, as I said, this API is experimental and doesn't properly works everywhere: https://developer.mozilla.org/en-US/docs/Web/API/Web_Bluetooth_API#browser_compatibility

Wohlstand, 2024-04-04:
Actually, this feature is only works when enable Experimental Web platform features flag:
[image]

[image]

Wohlstand, 2024-04-04:
Additionally, the Web Bluetooth API gets been refused because of security risks and complication of maintenance (including the blacklist of vulnerable/faulty devices).

Schokobecher, 2024-04-04:
> Actually, this feature is only works when enable Experimental Web platform features flag:
Can't confirm that:
[image]

BLE Transfer and DFU work fine with my 2019 MBP

Wohlstand, 2024-04-04:
Vivaldi is not fully Chromium, its way reworked including several features. On my end is a pure unmodified Chromium of the super latest version.

Schokobecher, 2024-04-04:
With a freshly installed Google Chrome (Stable, v123.0.6312.107)
[image]
Fresh Profile, never installed on this machine, this is not a Chrome/Chromium issue

deekue, 2024-06-08:
FWIW, I've had some success on Linux

Ubuntu 22.04.4
Google Chrome 125.0.6422.141 (with Experimental Web Platform features enabled)
[allmiibo LCD](https://www.aliexpress.com/item/1005006420542854.html) 2.10.2

firmware update worked via
https://thegecko.github.io/web-bluetooth-dfu/examples/web.html

had some issues with BLE File Transfer via
https://pixl.amiibo.xyz
enable 'BLE File Transfer' on the device, then click the 'Connect' button on the webiste, the pop-up window spins and scanning shows 'No compatible devices found'.

I went to the Bluetooth control panel and saw that a 'Pixl.js' device was listed as paired but not connected.  Removed the pairing and then the scanning worked and it connected fine.