solosky/pixl.js #224: Any way to go back to the previous amiibo selection folder?

Vorffeed, 2024-03-25:
Let me get this straight.
Is there any way to return to the previous folder to select another group of amiibo without having to click on the amiibo to enter the menu to generate random id to have to scale from there?
If there isn't, couldn't it be done in a way that pressing for an extended period 5-6s would take you back to the previous folder or failing that to the root folder database or amiibo emulator depending on where you are?
Without having to enter into the configuration menu of that amiibo in which you are?

solosky, 2024-03-25:
if you are using Amiibo Emulator and in a folder, you can select ".." (the first item in the list) to go up folder.
if you are using Amiibo Database, you can select "[Back]" to go back previous menu item. 

The button logic works nature and uniform, press/long press middle button to enter context menu, left/right button to select item. I will not take your suggestion to make long press as back action.

solosky, 2024-03-25:
As I said, the long press  action of middle button is used for context menu in may senarios, I am not going to change that unified interactive logic, sorry for that.

JostenSyon, 2024-04-05:
To be honest, I’m attempting to implement the ‘Back’ function as a ‘hold left’ command, but this can’t be used in certain scenarios where ‘hold left’ is required for its normal use, such as scrolling back through text while displaying the details of an amiibo. I can code this as a Proof of Concept; however, all commands should remain consistent, and this inconsistency is problematic as pointed out by the author

Vorffeed, 2024-04-07:
Doubt.
Are you saying that it is impossible to implement that, because there is conflict with other actions and adding a return behind the last amiibo in that folder could not be done either, as it appears in the menu above?

JostenSyon, 2024-04-10:
It's definitely a possibility. I'm currently working on this feature in my own branch. However, there's no estimated time for its completion, and ultimately, it's not my decision whether it will be merged into the main branch.