solosky/pixl.js #225: I can't delete or rename folder that i create

Biztone, 2024-03-25:
I add new folder to add amiibo but then i can't delete or rename folder. Please help me. When i try to delete it said "Failed to Delete File! [1]"

solosky, 2024-03-26:
Related issue #219 

The only way to solve this issue is formatting the whole flash, sorry for that.

Biztone, 2024-04-02:
> Related issue #219
> 
> The only way to solve this issue is formatting the whole flash, sorry for that.

thank you very much

xndvaz, 2026-04-25:
Closing this as a duplicate/already-answered case of #219.

The original issue was caused by invalid filenames created before filename validation was added. Existing invalid entries already written to flash still need storage formatting to recover, as noted in #219.

While checking this, I noticed a separate LFS-only directory rename bug: `vfs_lfs_rename_dir()` ignored the requested destination path and effectively renamed a directory to itself. That does not appear to be the root cause here because the current OLED/LCD board configs use SPIFFS, but I opened #439 to fix that separate bug.