solosky/pixl.js #228: DIY compilation of the FW according to the instructions from the documents section and a confusing result of it.

MayanKoyote, 2024-03-29:
If to compile FW according to the instructions '[03-Build-Firmware.md](https://github.com/solosky/pixl.js/blob/main/docs/en/03-Build-Firmware.md)', then only one file 'pixjs_ota_v1.zip' appears in the 'fw/_build' directory. 
And we have two different devices: LCD and OLED, each require different FW.
Since compiled this way FW 'blinds' my LCD device upon flashing, I suspect that it is for the OLED devices.

Anyway, the compiled 'ota' file is the only one and there are not two different 'ota' files for two different HW (screen) variants of devices.
[image]

So:
How to get both FW versions been compiled: for LCD and for OLED?
Or how to get only exactly LCD FW version been compiled?

solosky, 2024-03-29:
Good point, I will update the repository and document to help you build your firmware later.

solosky, 2024-03-29:
I have updated the https://github.com/solosky/pixl.js/blob/main/docs/en/03-Build-Firmware.md 
Please pull commits from main branch before building.

closed 2024-03-29 by MayanKoyote