solosky/pixl.js #234: Doubt about importing your amiibos

Vorffeed, 2024-04-01:
I am referring to the OLED version
Is there any way to access the storage through the usb connector, instead of web and bluetooth?
Because it only lets me transfer four at a time my amiibo collection without crashing the page and not being able to transfer I have to disconnect and re-pair the device and that I leave it on so it does not turn off > <
In addition to not being able to send folders from Buetooth with your amiibo configuration tucked in each of the folders, you have to create them one by one hand from the web

Vorffeed, 2024-04-02:
Ok, Thnaks, although it would have been nice to be able to access from the usb to transfer the files.

Schokobecher, 2024-04-03:
@Vorffeed uploading files via iNFC (iOS) is very reliable, I easily uploaded 500+ files in total without fail (separate folders with <100 files each). If you have an Android, give MToolsBLE a go.