solosky/pixl.js #236: The 'amibo' application automatically adds two-dot ellipsis to (some) text lines.

MayanKoyote, 2024-04-02:
For most menu item text lines, the ellipsis is specified in the localization file.
For example, like here:
https://github.com/solosky/pixl.js/blob/223609b07aa2b97a6adfeb8b2932985d1c4d3e64/fw/application/src/i18n/pt_PT.c#L71
But in two cases (same app, 'amiibo'), the application itself adds a two-dot ellipsis instead:
https://github.com/solosky/pixl.js/blob/223609b07aa2b97a6adfeb8b2932985d1c4d3e64/fw/application/src/app/amiibo/scene/amiibo_scene_file_browser_menu.c#L301
https://github.com/solosky/pixl.js/blob/223609b07aa2b97a6adfeb8b2932985d1c4d3e64/fw/application/src/app/amiibo/scene/amiibo_scene_file_browser_menu.c#L303

Is it possible to put this in a uniform order (i.e. the ellipsis is specified in the localization file)?

solosky, 2024-04-04:
> I would vote for putting the ellipsis inside the i18n.csv to unify all text as much as possible. In addition to that, ellipsis are typically three, not two dots. I already started working on reworking the de_DE translation, so I could integrate this across the board.

aggreed.

Schokobecher, 2024-04-06:
`_L_DELETE` is also used in the context of
https://github.com/solosky/pixl.js/blob/223609b07aa2b97a6adfeb8b2932985d1c4d3e64/fw/application/src/app/amiibo/scene/amiibo_scene_file_browser_menu.c#L258

Which would to look like this after swapping over the ellipsis to `i18n.csv`

> Delete... *FILENAME* ?

Is that acceptable? Or do we need another string in i18n.csv?

We also should be able to move the various question marks with something like this:

```c
snprintf(msg, sizeof(msg), "%s %s", getLangString(_L_DELETE), string_get_cstr(app->current_file));
```
```csv
_L_DELETE,Delete... %s ?
```
right?

MayanKoyote, 2024-04-06:
Be that as it may, but something needs to be done about it.

PR #240 (2024-04-07): Move punctuation to i18n.csv
This should close #236 

Changes:

- Moved hard-coded punctuation to i18n.csv
- Three dots instead of two (ellipsis typically have three)
- Added another string called `_L_DELETE_FILE` for use in `FILE_BROWSER_MENU_REMOVE_FOLDER`
- Updated the translations for `_L_DELETE_FILE` accordingly, please check if zh_Hans and zh_TW are useable
- Rebuilt language files

I compiled and flashed this version, and English/German look correct. Let me know if anything needs to be reworked, or I missed anything!
closed 2024-04-13 by MayanKoyote