solosky/pixl.js #237: Usage of both _L_ON/_L_OFF and _L_ON_F/_L_OFF_F to achieve a similar result - to display ON/OFF in square brackets.

MayanKoyote, 2024-04-03:
In general, to display the [ON]/[OFF] states on the screen, special lines are used that are defined in the localization file as _L_ON_F/_L_OFF_F.
https://github.com/solosky/pixl.js/blob/223609b07aa2b97a6adfeb8b2932985d1c4d3e64/fw/application/src/i18n/en_US.c#L5C1-L6C26

But in some cases, the values of the _L_ON/_L_OFF lines of the localization file are used instead. And they are automatically placed in square brackets by the applications itself.
(I think) these are the cases (although maybe I'm mistaking):
1. pixl.js/fw/application/src/app/amiidb/scene/amiidb_scene_amiibo_detail_menu.c
https://github.com/solosky/pixl.js/blob/223609b07aa2b97a6adfeb8b2932985d1c4d3e64/fw/application/src/app/amiidb/scene/amiidb_scene_amiibo_detail_menu.c#L99
https://github.com/solosky/pixl.js/blob/223609b07aa2b97a6adfeb8b2932985d1c4d3e64/fw/application/src/app/amiidb/scene/amiidb_scene_amiibo_detail_menu.c#L111
https://github.com/solosky/pixl.js/blob/223609b07aa2b97a6adfeb8b2932985d1c4d3e64/fw/application/src/app/amiidb/scene/amiidb_scene_amiibo_detail_menu.c#L146

2. pixl.js/fw/application/src/app/amiibolink/scene/amiibolink_scene_menu.c
https://github.com/solosky/pixl.js/blob/223609b07aa2b97a6adfeb8b2932985d1c4d3e64/fw/application/src/app/amiibolink/scene/amiibolink_scene_menu.c#L68
https://github.com/solosky/pixl.js/blob/223609b07aa2b97a6adfeb8b2932985d1c4d3e64/fw/application/src/app/amiibolink/scene/amiibolink_scene_menu.c#L97

3. pixl.js/fw/application/src/app/amiibo/scene/amiibo_scene_amiibo_detail_menu.c
https://github.com/solosky/pixl.js/blob/223609b07aa2b97a6adfeb8b2932985d1c4d3e64/fw/application/src/app/amiibo/scene/amiibo_scene_amiibo_detail_menu.c#L132
https://github.com/solosky/pixl.js/blob/223609b07aa2b97a6adfeb8b2932985d1c4d3e64/fw/application/src/app/amiibo/scene/amiibo_scene_amiibo_detail_menu.c#L168

4. pixl.js/fw/application/src/app/settings/scene/settings_scene_main.c
https://github.com/solosky/pixl.js/blob/223609b07aa2b97a6adfeb8b2932985d1c4d3e64/fw/application/src/app/settings/scene/settings_scene_main.c#L172
___
Except these cases of usage of the _L_ON/_L_OFF lines as substitutes for _L_ON_F/_L_OFF_F, the _L_ON/_L_OFF lines are used as follows:
there is usage of _L_OFF on its own (w/o brackets):
pixl.js/fw/application/src/app/settings/scene/settings_scene_sleep_timeout.c
https://github.com/solosky/pixl.js/blob/223609b07aa2b97a6adfeb8b2932985d1c4d3e64/fw/application/src/app/settings/scene/settings_scene_sleep_timeout.c#L68
and not a single use of _L_ON on its own (w/o brackets).

(At least I was unable to find anything else using github's search.)
___

So, is it possible to use only the lines defined in the localization file as _L_ON_F/_L_OFF_F to display the [ON]/[OFF] states on the screen?

Also, is it possible to replace them (as one of the options) with pseudo-graphic symbols, for example, such as ☑ (U+2611) and ☒ (U+2612)?
This would return the displaying of the [ON]/[OFF] states to a single character (开/关) and reduce onscreen text "jumping" when switching states. (Yes, I know that in some languages they have the exact opposite meaning. But this does not prevent speakers of such languages from successfully using imported tech, websites, etc. In addition, this can be easily redefined in the same localization file.)

Schokobecher, 2024-04-03:
> Also, is it possible to replace them (as one of the options) with pseudo-graphic symbols, for example, such as ☑ (U+2611) and ☒ (U+2612)?

`wenquanyi_9pt_u8g2.bdf` only includes

> COMMENT WenQuanYi bitmap fonts include all 20,932 Unicode 5.2 
> COMMENT CJK Unified Ideographs (U4E00 - U9FA5) and 6,582 
> COMMENT CJK Extension A characters (U3400 - U4DB5) at 
> COMMENT 5 different pixel sizes (9pt-12X12, 10pt-13X13, 
> COMMENT 10.5pt-14x14, 11pt-15X15 and 12pt-16x16 pixel).
> COMMENT Use of this bitmap font for on-screen display of Chinese 
> COMMENT (traditional and simplified) in web pages and elsewhere 
> COMMENT eliminates the annoying "blurring" problems caused by 
> COMMENT insufficient "hinting" of anti-aliased vector CJK fonts. 
> COMMENT In addition, Latin characters, Japanese Kanas and 
> COMMENT Korean Hangul glyphs (U+AC00~U+D7A3) are also included.

So no glyphs for universal icons are included

MayanKoyote, 2024-04-03:
It's always easy to replace or add some symbols to the font.
And, as I understand it, the font license allows to do this.

Schokobecher, 2024-04-03:
Actually scratch that, since `u8g2_font_siji_t_6x10` already provides some symbols:
[image]

But not sure any of these are very suited for this use-case

MayanKoyote, 2024-04-03:
The places intended for the symbols (U+2611) and (U+2612) in the font are currently not populated, so it's possible fill them and it turns out something like this:
[image]
[image]

MayanKoyote, 2024-04-05:
> Since we already have the square brackets around the icon, I suggest to remove the border of the ON / OFF icon.

U+2713 and U+2717 (although perhaps they should be placed in places U+2714 and U+2718):
[image]
[image]

closed 2024-06-03 by MayanKoyote