solosky/pixl.js #241: How should the [More..] button of the 'Search' function of the 'Amiibo Database' app work if there are more than 100 search results?

MayanKoyote, 2024-04-07:
The 'Amiibo Database' app has a 'Search' function.
The number of search results displayed per page is determined by the variable 'LIST_VIEW_ITEM_MAX_COUNT', it is equal to 100.
(Since my device came to me from Aliexpres, it contains a lot of material for testing.)
I do search for the letter 'b' and get information that there are 186 results in total and a list of the first 100 results.
Next comes the [More..] button. It is logical to assume that it should show me the remaining 86 results.
But I press it, and... nothing happens, the list of results does not change.

Am I misunderstanding something or is something not working the way it should?

solosky, 2024-04-07:
The [More..] only indicates that there are more that 100 items of the results, but due to the limitation of MCU memory, the device only could display 100 results, please add more characters to narrow down your search results.

solosky, 2024-04-07:
Considering display more search results is not make sense because it's hard to go through all search results in such small display, I do not add logic as your suggestion.

MayanKoyote, 2024-04-07:
It's clear now.
Then it might be worth renaming '[More..]' into something like '...and so on.' and removing the brackets around it?

solosky, 2024-04-07:
I do suggest that showing a message box displayed "Too many search results. \n Try update search criteria." when user click the [More..] item.

closed 2024-04-09 by MayanKoyote