solosky/pixl.js #242: The device does not want to show characters from the line [_L_APP_CHAMELEON_CARD_SET_NICK_FAILED] that are not Basic Latin.

MayanKoyote, 2024-04-07:
en_US - line is 'Set Nick Failed!' and the message is displayed as expected: 'Set Nick Failed!':
[image]

zh_Hans - line is '设置卡名失败' and the message displayed is ' '. Everything is missing, empty message:
[image]

fr_FR - line is 'Définir le Pseudo a Échoué!' and the message displayed is 'Dfinir le Pseudo a chou!'. All the characters 'é' and 'É' are missing:
[image]

ja_JP - line is 'ニックネームが設定できませんでした!' and the message displayed is '!'. Everything except the exclamation mark is missing:
[image]

It seems that the app wants to display only Basic Latin characters and does not want to display any others.

A similar story is observed in the case involving [_L_APP_CHAMELEON_CARD_ADV_ID_EDIT_INVALID_INPUT] line.

solosky, 2024-04-07:
Quickly checked, the mui_toast_view does not set wqy font before draw characters. 
I will fix this later today.

solosky, 2024-04-07:
I should fix this issue, but it comes a new issue: The text is out of border when toast content displays more than one line.
I will fix that when I am avaliable.

commit 198007d2 (2024-04-07): fixed #242: not set correct font when toast draw message
closed 2024-04-12 by solosky