solosky/pixl.js #253: Device won't power on after last fw update

Samoth980, 2024-05-13:
Hi all,
I've received today my device, I've followed the instruction reported here but after the fw update the device won't turn on, suggestion?

solosky, 2024-05-13:
see https://github.com/solosky/pixl.js/blob/main/docs/en/02-Flash-Firmware.md#repair-wrong-firmware-version

Samoth980, 2024-05-13:
i saw it but there is no way to make it visible over the browser as before also the combination of left, middle... doesn't produce any blue light, it work only if I click on right (x3) but the browsed won't recognize (an external app yes,

solosky, 2024-05-13:
try use nrf connect app to upgrade, it is more stable when upgrading firmware.

Samoth980, 2024-05-13:
Is not clear, because with the last release the sequence button for the bt mode is different but i've solved.

impeeza, 2024-05-13:
Which device do you have?

Samoth980, 2024-05-14:
the LCD version, i've installed the last version and than crashed. Maybe a check about the hardware before forcing the upgrade will be useful

solosky, 2024-05-16:
> the LCD version, i've installed the last version and than crashed. Maybe a check about the hardware before forcing the upgrade will be useful

Good idea, let think how to do with this.

impeeza, 2024-05-19:
> the LCD version, i've installed the last version and than crashed. Maybe a check about the hardware before forcing the upgrade will be useful

can you try changing the battery?  mine last only few uses.

Samoth980, 2024-05-19:
I've solved, the issue was due to the wrong installed sw pushed and the
reset sequence reported into the documentation doesn't works with the last
release.

impeeza, 2024-05-19:
Thanks.  for let us to know.

Samoth980, 2024-05-19:
How can I change the battery?It is soldered onto the motherboard. Also how
this can solve a software incompatibly issue?

impeeza, 2024-05-20:
> How can I change the battery?It is soldered onto the motherboard. Also how this can solve a software incompatibly issue? Il dom 19 mag 2024, 07:31 impeeza ***@***.***> ha scritto:
> […](#)

Ah, ok,  some LCD have a CR2032 battery.

Yours is the latest model which have the lion battery.  

if your device have the back light set up and you don't see it may be a depleted battery even with wrong fimware.   

ON LATEST FIRMWARE, you need to press:

-Any key to wake up the device
-Left
-Enter (middle button)
-Left 4 times
-Enter (middle button)

TO PUT YOUR DEVICE ON FIRMWARE UPGRADE MODE.

https://github.com/solosky/pixl.js/assets/41979604/b8687d65-e491-4236-ac06-f4667ec56aa3

solosky, 2024-07-29:
closed as issue resolved.

closed 2024-07-29 by solosky