solosky/pixl.js #260: contents of u8g2_font_wqy12_t_gb2312a.c

impeeza, 2024-08-11:
Hello there,

Preparing the https://github.com/solosky/pixl.js/pull/259 I noted what the original `fw/application/src/mui/u8g2_font_wqy12_t_gb2312a.c` have the header `#ifdef U8G2_USE_LARGE_FONTS` 

But the same file generated by `fw/scripts/font_data_gen.sh` have the header: `#ifdef U8G2_USE_LARGE_GB2312_FONT`

What is the difference between those two fonts?  should the script `fw/scripts/font_data_gen.sh` be updated to use the new font?

I am attaching a original file from github repo and a generated one.

regards.

[u8g2_font_wqy12_t_gb2312a_Generated.zip](https://github.com/user-attachments/files/16576121/u8g2_font_wqy12_t_gb2312a_Generated.zip)
[u8g2_font_wqy12_t_gb2312a_Original.zip](https://github.com/user-attachments/files/16576122/u8g2_font_wqy12_t_gb2312a_Original.zip)

solosky, 2024-08-12:
Hi, the script ``` fw/scripts/font_data_gen.sh ``` is deprecated with python script ```font_data_gen.py```, the new python script could be run in Windows/Linux/Macos. 
The macro define ```U8G2_USE_LARGE_FONTS``` is generated by ```bdf_conv```, I adopt the define in Makefile. 
```
ifneq ($(RELEASE), 1)
CFLAGS += -DDEBUG
CFLAGS += -DEBUG_NRF
#CFLAGS += -DU8G2_USE_LARGE_FONTS
else
CFLAGS += -DNDEBUG
CFLAGS += -DU8G2_USE_LARGE_FONTS
```

impeeza, 2024-08-12:
Thanks a lot for your quick response,  trying just now. and taking the correspondent changes to the last PR.

closed 2024-08-12 by impeeza

impeeza, 2024-08-12:
Updated the Pull request with the correct files.  thanks a lot.

commit c23fe4ae (2024-08-12): Fixed #260: remove deprecated script font_data_gen.sh