solosky/pixl.js #265: Where do I get the firmware zip for allmiibo????

Breadmasterpro, 2024-08-26:
I can not find the firmware zip package mentioned in 02-flash-firmware.md to update my allmiibo with web method. where is it!?!?

solosky, 2024-08-26:
please download from release page. if you are going to use OTA to update firmware, unzip it and use xxx_ota.zip to upgrade.

Breadmasterpro, 2024-08-26:
> please download from release page. if you are going to use OTA to update firmware, unzip it and use xxx_ota.zip to upgrade.

my appolagies, I cannot seem to find it! 

Is this it?
[image]

Breadmasterpro, 2024-08-26:
> please download from release page. if you are going to use OTA to update firmware, unzip it and use xxx_ota.zip to upgrade.

AHA! found it! thanks! 
[image]

Breadmasterpro, 2024-08-26:
> please download from release page. if you are going to use OTA to update firmware, unzip it and use xxx_ota.zip to upgrade.

It Freezes at this point. does not move after 5 minutes. even after retrying.
[image]

Breadmasterpro, 2024-08-26:
now my allmiibo is stuck on 'DFU update waiting... ' HELP!

Breadmasterpro, 2024-08-26:
WHEW I FIXED IT!!! yes!! new firmware is very cool! I fixed it by using the nrf connect app. all fixed thankyou!!!

Breadmasterpro, 2024-08-26:
it keeps getting stuck on whitescreen...

closed 2024-08-27 by Breadmasterpro

solosky, 2024-08-27:
Hi, just want to know, have you resolved your issue ?