solosky/pixl.js #266: Bootloop in the DFU Updater mode

GlebDGT11, 2024-08-27:
Hi, @solosky , I have recently catch problem for my allmiibo, that I bought on 2nd of July and getting it in 29th of July. So, the problem: When I wanted to download latest firmware to 2.11.0 from 2.10.2, firstly screen looking like it's getting hammered, but not, I quickly put it into DFU Updater to install that firmware again, and while updating is getting finished, it just stops at 240k firmware bytes and started bootlooping in the DFU Updater mode, and it won't able to return to Main Menu

Uploading VID_2024_08_27_16_45_14_1262661556606660999115.mp4…

solosky, 2024-08-28:
Sorry, I cannot see the video you posted here. 
you can try upgrade firmware with MTools BLE and NRF Connect. 

https://github.com/solosky/pixl.js/blob/main/docs/en/02-Flash-Firmware.md#nRF-Connect-APP

GlebDGT11, 2024-08-28:
I fix the problem, and it's now working, thanks to my friend that helped me out from this

impeeza, 2024-08-28:
How did you solve the problem?   please close the issue if is solved.

GlebDGT11, 2024-08-28:
I just short circuited some contacts on Allmiibo Board to reset the system and then writing the firmware again

closed 2024-08-28 by GlebDGT11