solosky/pixl.js #267: After updating the strip instead of the text, can I fix it?

A5littleangel, 2024-08-27:
I updated with infc to the latest version, but after updating, there are only stripes on the screen, can I fix this?
[image]

impeeza, 2024-08-27:
you apply the OLED firmware to a LCD unit.

Follow the instructions to apply the firmware update blindly

solosky, 2024-08-28:
Instruction for repair wrong firmware: 

https://github.com/solosky/pixl.js/blob/main/docs/en/02-Flash-Firmware.md#repair-wrong-firmware-version

A5littleangel, 2024-08-28:
> Instruction for repair wrong firmware:
> 
> https://github.com/solosky/pixl.js/blob/main/docs/en/02-Flash-Firmware.md#repair-wrong-firmware-version

Where to press at this moment are these keys (X N)?

LEFT X N

The strip on the screen does not disappear after 45 seconds either
[image]

impeeza, 2024-08-28:
«LEFT x N» (times)  is

If the firmware version before 2.11.x, press LEFT x 4 If the firmware version after 2.11.x, press LEFT x 5

is on the instructions,  that means if the wrong firmware you flash  is below 2.11  you need to press LEFT 4 times,  if the firmware is newer your need to press LEFT 5 times.

impeeza, 2024-08-28:
And yes is normal the strip do not disappear because the driver is wrong.

A5littleangel, 2024-08-28:
> And yes is normal the strip do not disappear because the driver is wrong.

Thanks for the previous answer, I've been doing everything according to the instructions for an hour now, writing this 4096/357604 firmware bytes written
What to do? How much should be updated approximately?

[image]
[image]

impeeza, 2024-08-28:
Send the update again sometimes it can hang, if you start again from the app or browser the upgrade will resume.

A5littleangel, 2024-08-28:
> Send the update again sometimes it can hang, if you start again from the app or browser the upgrade will resume.

I tried it 20 times, but it doesn't go beyond that.
4096/357604 firmware bytes written

I'm trying through this
https://thegecko.github.io/web-bluetooth-dfu

through https://pixl.amiibo.xyz /
does not find the device

impeeza, 2024-08-28:
trY https://github.com/solosky/pixl.js/blob/main/docs/en/02-Flash-Firmware.md#nRF-Connect-APP

MayanKoyote, 2024-08-29:
Are you flashing from a PC? Try flashing from an Android phone.

A5littleangel, 2024-08-29:
Thank you, it turned out to flash through the PC, it also did not work through the phone, it just does not see the device.

When emulating amiibo, the screen freezes and nothing happens, can I reboot somehow? Already turned on, and does not turn off for 7 hours

impeeza, 2024-08-29:
on your model you can remove the battery to full reboot the device,  also on settings menu there is an option to reboot the device.