solosky/pixl.js #270: Suggestion: Is it possible to let user enter their own UUID of choice?

t0mizwf26, 2024-09-04:
For "Amiibo Emulator", in addition to "Rand. Tag", is it possible to implement a new feature to let user enter their own UUID?

Something like:
```
Rand. Tag
Auto Rand. [OFF]
{new} Cust. UUID
{new} Use C.UUID [ON]
...
```
Use Case Scenario (example):

**TL;DR** By entering/cloning real Amiibo UID into pixl.js, enables user to transfer Amiibo data back and forth between real Amiibo and pixl.js

1. User obtain UID from a real Amiibo (Mario for example) using smartphones app etc.
2. User enter that UID into pixl.js device (Mario.bin)
3. User playing SSBU on Switch at home using real Amiibo and train the Amiibo FP
4. Before traveling, user let SSBU read real Amiibo's data and then write the data into pixl.js (with same UUID)
5. When traveling, user continue to train the FP using pixl.js
6. When back home, user let SSBU read pixl.js data and write the data back into real Amiibo

solosky, 2024-09-06:
Hi, thank your suggestion. 
I don't think this feature is nessary, you can archive same target by steps bellow:

* Get a fully dump from your amiibo by NFC reader app of Mobile Phone (Or search the dump file from internet)
* Put the amiibo bin file into Pixl.js 
* Select the amiibo bin file, and turn off the Auto Rand. function, so the id of amiibo bin file do not change
* when travelling, user continue to train the FP using pixl.js, the data will write to amiibo bin file
*  When back home, user let SSBU read pixl.js data and write the data back into real Amiibo

t0mizwf26, 2024-09-07:
Thanks so much for the explanation, but I guess this feature might still be useful if someone wants to transfer existing save data from pixl.js to a newly purchased Amiibo?

For example:
1. User doesn't have a real Mario Amiibo, so user is using pixl.js Mario with UID "XXXX" to play Smash Bros
2. User bought a new real Mario Amiibo (UID "YYYY") and wants to transfer / copy the Smash Bros Amiibo FP save data from pixl.js Mario to real Mario Amiibo

I heard Tagmo for Android can "Ignore Tag ID #" so it can force update any Amiibo even they have different UID, so it can write UID "XXXX" save data into UID "YYYY", but the problem is Tagmo not available on iOS. 

If we can have this "custom UUID" feature implemented, maybe people will then able to transfer save data between any Amiibo without using any special phone app? (except they still need to find out their Amiibo's UID)

Breadmasterpro, 2024-09-18:
> Hi, thank your suggestion. I don't think this feature is nessary, you can archive same target by steps bellow:
>
> * Get a fully dump from your amiibo by NFC reader app of Mobile Phone (Or search the [...]

I think it would be a good feature, and if you could save the amiibo uuid's in case of a reset that would be even better!

commit a1965c7e (2024-10-13): Fixed #270: add option to custom uid
closed 2024-10-13 by solosky