solosky/pixl.js #273: BUG: 2.11.1 new Amiibo wrong ID

t0mizwf26, 2024-09-05:
According to AmiiboAPI https://github.com/N3evin/AmiiboAPI/issues/181

The IDs in PR#272

> 0801000004360402,Callie - Alterna,小拟 - 幻界
> 0802020004370402,Marie - Alterna,小萤 - 幻界
> 0803010004380402,Pearl - Side Order,小姬 - 秩序篇
> 0804010004390402,Marina - Side Order,饭田 - 秩序篇

are wrong. They are trying to re-verify the IDs ATM.

AbandonedCart, 2024-09-05:
https://github.com/N3evin/AmiiboAPI/issues/181#issuecomment-2325174588 are valid.

solosky, 2024-09-06:
Confirmed from users, the amiibo ids are correct.

AbandonedCart, 2024-09-06:
> Confirmed from users, the amiibo ids are correct.

The set I posted are the official ones retrieved from retail statues. The set in the first post have a typo that circulated ahead of the release.

solosky, 2024-09-06:
I just checked the Amiibo API data, you are correct, I think I will update the amiibo ID in next release to sync up with Amiibo API. 

Thanks.

commit 71f4156c (2024-09-25): fixed #273: Fix Splatoon 3 Alterna and Side Order
closed 2024-10-13 by solosky