solosky/pixl.js #274: Allmiibo screen randomy starts bugging out

Breadmasterpro, 2024-09-08:
After updating to v2.11.0 the screen will sometimes go completely blank or bug out and go upside down. All the allmiibo functions still work when the screen goes blank and I can still scan and switch amiibo but I just can't see anything at all. The only way to fix this is to let it enter sleep mode then turn it on again.

https://github.com/user-attachments/assets/f10a9185-2c19-4424-a983-24fd577ae5cf

[image]
[image]

Breadmasterpro, 2024-09-08:
[image]

Breadmasterpro, 2024-09-08:
[image]

solosky, 2024-09-08:
This issue happens before, still can not know the reason, try to flash firmware again or reset the device by pull out the battery.

Breadmasterpro, 2024-09-09:
> This issue happens before, still can not know the reason, try to flash firmware again or reset the device by pull out the battery.

Ok

closed 2024-09-16 by Breadmasterpro