solosky/pixl.js #278: Flashing firmware with webpage dfu just freezez my allmiibo.

Breadmasterpro, 2024-09-14:
https://github.com/user-attachments/assets/7e354a5a-17ac-41bb-ab4f-8fc38e138b9d

Here as you can see where it freezes. after this it does nothing, i have waited 30 mins and nothing changed, and my allmiibo was stuck in dfu mode. I had to flash using nrf connect but the almiibos in browser dont save so it is almost unusable.
(i have sped up the video so it fits)

MayanKoyote, 2024-09-15:
bluetooth module is USB whistle? Is it completely shitty one?

Breadmasterpro, 2024-09-15:
> bluetooth module is USB whistle? Is it completely shitty one?

It is a normal allmiibo

solosky, 2024-09-16:
The web ble is not stable currently, please use NRFConnect or MTools BLE to update firmware. 
Instruction: https://github.com/solosky/pixl.js/blob/main/docs/en/02-Flash-Firmware.md

Breadmasterpro, 2024-09-16:
> The web ble is not stable currently, please use NRFConnect or MTools BLE to update firmware. Instruction: https://github.com/solosky/pixl.js/blob/main/docs/en/02-Flash-Firmware.md

When I use nrf connect to flash the firmware the amiibos don't save after it enters sleep mode. While it is still on it saves but once it enters sleep mode it changes. I don't know if that is a bug or if it is nrf connect app.

closed 2024-09-18 by Breadmasterpro