solosky/pixl.js #279: Amiibo in 'amiibo database' do not save

Breadmasterpro, 2024-09-18:
The amiibo in 'amiibo emulator' save but the amiibos in 'amiibo emulator' are not up to date because some are missing and I prefer to use the database anyway. The amiibos in 'amiibo database' reset everything the divice goes to sleep. While it is still on they remain the same but once the divice enters sleep state and then I turn it on and go back to the same amiibo, it has reset. I have made sure the 'randomise' option is off, I have rebooted the allmiibo, I have reset settings and I have reflashed the firmware many times but nothing fixes the issue, it is sad because I did a ton of battles is splatoon but the amiibos aren't saving so I could not get the amiibo gear :( can you please look into this issue?

Breadmasterpro, 2024-09-18:
> The amiibo in 'amiibo emulator' save but the amiibos in 'amiibo emulator' are not up to date because some are missing and I prefer to use the database anyway. The amiibos in 'amiibo database' reset [...]

Update: it also resets if I just change amiibo then go back to the same on without the divice entering sleep mode.

fedprod, 2024-09-20:
That is because the Database is only as a reference and not for personal Amiibos.
For that you have to SAVE them as an Amiibo in your personal list!

solosky, 2024-09-21:
Yeah, if you ```go browse..``` menu and select one of amiibo in Amiibo Database, the amiibo data is volatile, means that it is generated amiibo data on the fly for one time use and do not save it to flash. 
If you want to write data to the generated amiibo, first you need to go to the context menu and select `save..`  to one slot. 
once you want to write data to the amiibo, you need first select ``` My Amiibo``` and select slot that you want to read and save to, the amiibo will be saved automatically when any write to the amiibo.

Breadmasterpro, 2024-09-21:
> Yeah, if you go `browse..` menu and select one of amiibo in Amiibo Database, the amiibo data is volatile, means that it is generated amiibo data on the fly for one time use and do not save it to [...]

Oh thanks, can you add an option to have my amiibo on the main menu? Maybe make it replace amiibolink because that never works. Are the amiibo in amiibo emulator supposed to be up to date or are they like references? Thanks for your help! I'll go save my amiibo now!

solosky, 2024-09-21:
If you have amiibo data files, you can upload to device and use Amiibo Emulator to emulate amiibo data files.
If you don't have amiibo data files, you can use Amiibo Database to generate Amiibo data file on the fly, and save it into slot if you would like to use as personal amiibo data files.

I am not going to add My Amiibo to the main menu, because that is not following the design pattern of Pixl.js firmware.