solosky/pixl.js #286: DFU Android app says that Allmiibo doesn't support nRF5 DFU

Splatboy0612, 2024-10-14:
I used DFU app by Nordic Semiconductor to update my Allmiibo to 2.10.2 when I got it for the first time in April, now it won't let me, I would use the official web-site, but my first experience with it gave me a bricked Allmiibo until I updated it with DFU app

MayanKoyote, 2024-10-15:
This is redundant.

Although if you look into it, this is what is currently contained in the 'Settings' section:

= Settings =
 - Version - only information
 - Language - setting
 - External Storage - information and formatting action. Well, it fits the definition of "settings", since it's set (format) storage. Although it (formatting) can also be attributed to "management".
 - Backlight - setting
 - Menu Animation - setting
 - LiPO Battery - setting
 - Memory Used - setting and information
 - Fast Wakeup - setting
 - Sleep Timeout - setting
 - Go Sleep - action. Definitely not a setting
 - Firmware Update - action. Very indirectly related to the settings (only since it's about firmware update)
 - System Reboot - action. Definitely not a setting
 - Reset Default Settings - action. But directly related to the settings
 - About Device - only information

From what is directly or indirectly not related to the settings, it's possible to compose a new separate section, let's call it 'Management':

= Management =
 - Go Sleep - action
 - System Reboot - action
 - Firmware Update - action

and if we transfer 'Firmware Update' action here, then perhaps it can also be moved here (to 'Management' section):

 - Version - only information
 - About Device - only information

So it will turn out like this:
= Management =
 - About Device - only information
 - Version - only information
 - Firmware Update - action
 - System Reboot - action
 - Go Sleep - action

solosky, 2024-10-15:
> This is redundant.
>
> Although if you look into it, this is what is currently contained in the 'Settings' section:
>
> = Settings =
>
> * Version - only information
> * Language - setting
> * External Storage - [...]

Good point of view. 
The setttings menu sames a mess currently, I really don't give a deep thought like this. 
I may have re-structure the menu settings in future release.

Splatboy0612, 2024-10-22:
> This is redundant.
>
> Although if you look into it, this is what is currently contained in the 'Settings' section:
>
> = Settings =
>
> * Version - only information
> * Language - setting
> * External Storage - [...]

Oh, ok, then I'll close the case

closed 2024-10-22 by Splatboy0612