solosky/pixl.js #287: Something is wrong with the Português(Portugal) localization - it doesn't update with time.

MayanKoyote, 2024-10-14:
I have long noticed that the localization Português(Portugal) file "pt_PT.c" is not updated (time stamp) when executing the script "i18n_gen.py".
Today I took a closer look at it - and, at least for some time now, there also have been no newly added lines. File "pt_PT.c" now still contains only 169 lines instead of 183, as in the "en_US.c" file (or any other).
I also noticed that the i18n.csv file is completely missing a column titled "pt_PT". It seems like it was "lost" from there immediately after being added - introduced with commit 0ffcca0 and disappeared (as far as I understand) with commit f888bc8.
Is this a bug? Or is there some other story with it? (I don’t understand Portuguese at all)

MayanKoyote, 2024-10-15:
I restored pt_PT, rebuilt the firmware, tested it and in my opinion everything works as it should now.
If there are no objections, then I have an almost ready commit to restore the functionality of pt_PT.

MayanKoyote, 2024-10-15:
accept PR #285 first, otherwise I'm afraid that something will mix up badly again

solosky, 2024-10-15:
#285 is merged, thanks.

MayanKoyote, 2024-10-15:
Done. PR to restore the pt_PT suppport is PR #288.

closed 2024-10-16 by MayanKoyote