solosky/pixl.js #289: SSD1306 in I2C mode

Ahmadkorhani, 2024-11-03:
Can you implement SSD1306 in I2C mode please ?

solosky, 2024-11-07:
May I know why would need this feature?

Ahmadkorhani, 2024-11-07:
Thank you for your response, I can't find the spi version in my country, only I2C version.

I kept SPIM instance to 0 and enabled TWIM1 for oled, added the correct handler and callbacks (I tested everything in separate project using u8g2 library and my functions are working) but on my nrf52832 proto board the screen show only white noise (static white dots with random position on startup)

Ahmadkorhani, 2024-11-07:
I meant on separate projects my u8g2 I2C implementation was working, same code on pixl is showing white dots.
I double checked configuration in sdk_config.h

Ahmadkorhani, 2024-11-07:
here is the log from RTT:
00> <info> app_timer: RTC: initialized.
00> 
00> <debug> app: FLASH ID:
00> 
00> <debug> app:  EF 40 18               |.@.     
00> 
00> <info> app: W25Q128BV detection
00> 
00> <info> app: external fs mount: 0
00> 
00> <info> app: settings init: 8
00> 
00> <info> app: nrf_power_resetreas_get: 0x0000
00> 
00> <info> app: noinit area: [0x20002BD0, 0x20002FDC], 259 bytes
00> 
00> <info> app: m_cache_data address: 0x20002BD0
00> 
00> <info> app: Cleaning cache...
00> 
00> <info> app: Reset noinit ram done.
00> 
00> <info> app: WeakUp from button
00> 
00> <info> app: amiibo key read: -90
00> 
00> <info> app: init done
00> 
00> <info> app: Cache data: enabled = 0, id = 0
00> 
00> <info> app: wakeup reason: 0
00> 

but still nothing but white dots on the screen, pushing the buttons do nothing

solosky, 2024-11-07:
It's a lot of work to adopt new screen,  please post your modification of code here, otherwise I cannot give your any advice.

Ahmadkorhani, 2024-11-07:
Here is the modified mui_u8g2.c file . the original only initialize the oled in spi mode. this code is working perfectly on standalone project but not inside this pixl

`#include <stdio.h>

#include "app_error.h"
#include "nrf_delay.h"
#include "nrf_gpio.h"
#include "nrf_log.h"
#include "settings.h"

#include "u8g2.h"
#include "u8x8.h"

#include "app_pwm.h"

#include "boards_defines.h"
#include "nrf_drv_twi.h"

u8g2_t u8g2;

uint8_t m_u8g2_initialized = 0;
/* Indicates if operation has ended. */
static volatile bool m_xfer_done = false;
static bool readsomething = false;

#define TWI_ADDRESSES    127
#define OLED_ADDR        0x3C
#define TWI_INSTANCE_ID     1
//static const nrf_drv_twi_t m_twi = NRF_DRV_TWI_INSTANCE(TWI_INSTANCE_ID);
static const nrf_drv_twi_t m_twi = NRF_DRV_TWI_INSTANCE(1);

uint8_t u8g2_nrf_gpio_and_delay_twi_cb(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr);
uint8_t u8x8_HW_com_twi_nrf52832(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr);
void twi_init (void);
void twi_handler(nrf_drv_twi_evt_t const * p_event, void * p_context);

void twi_handler(nrf_drv_twi_evt_t const * p_event, void * p_context)
{
    switch(p_event->type)
    {
        case NRF_DRV_TWI_EVT_ADDRESS_NACK:
        {
            // NRF_LOG_ERROR("Got back NACK");
            m_xfer_done = true;            
            break;
        }
        case NRF_DRV_TWI_EVT_DATA_NACK:
        {
            // NRF_LOG_ERROR("Got back D NACK");
            m_xfer_done = true;            
            break;
        }        
        case NRF_DRV_TWI_EVT_DONE:
        {
            if (p_event->xfer_desc.type == NRF_DRV_TWI_XFER_RX)
            {
                readsomething=true;
            }
            m_xfer_done = true;
            break;
        }        
        default:
        {
            m_xfer_done = false;
            break;
        }
    }    
}

void twi_init (void)
{
    ret_code_t err_code;

    const nrf_drv_twi_config_t twi_oled_config = {
       .scl                = OLED_I2C_PIN_SCL,
       .sda                = OLED_I2C_PIN_SDA,
       .frequency          = NRF_DRV_TWI_FREQ_400K,
       .interrupt_priority = APP_IRQ_PRIORITY_HIGH,
       .clear_bus_init     = false,       
    };

    err_code = nrf_drv_twi_init(&m_twi, &twi_oled_config, twi_handler, NULL);    
    APP_ERROR_CHECK(err_code);

    nrf_drv_twi_enable(&m_twi);
}

uint8_t u8g2_nrf_gpio_and_delay_twi_cb(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
    switch(msg)
    {   
        case U8X8_MSG_DELAY_MILLI:
            // NRF_LOG_INFO("nrf_delay_ms(%d)", arg_int);
            nrf_delay_ms(arg_int);
            break;

        case U8X8_MSG_DELAY_10MICRO:
            // NRF_LOG_INFO("nrf_delay_us(%d)", 10*arg_int);
            nrf_delay_us(10*arg_int);
            break;
        
        default:
            u8x8_SetGPIOResult(u8x8, 1); // default return value
            break;  
    }
    return 1;
}

uint8_t u8x8_HW_com_twi_nrf52832(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
    uint8_t *data;
    bool res = false;
    ret_code_t err_code;    
    static uint8_t buffer[32];
    static uint8_t buf_idx;
    switch(msg)
    {
      case U8X8_MSG_BYTE_SEND:
      {
            data = (uint8_t *)arg_ptr;      
            while( arg_int > 0 )
            {
              buffer[buf_idx++] = *data;
              data++;
              arg_int--;
            }      
            break;  
      }      
      case U8X8_MSG_BYTE_START_TRANSFER:                    
      {
            buf_idx = 0;            
            m_xfer_done = false;
            break;
      }
      case U8X8_MSG_BYTE_END_TRANSFER:
      {            
            uint8_t addr = u8x8_GetI2CAddress(u8x8);
                       
            err_code = nrf_drv_twi_tx(&m_twi, u8x8_GetI2CAddress(u8x8) , buffer, buf_idx, false);
            APP_ERROR_CHECK(err_code);
            while (!m_xfer_done)
            {
                __WFE();
            }
            break;
      }
      default:
            return 0;
    }
    return 1;
}

void mui_u8g2_init(u8g2_t *p_u8g2) {

    twi_init();
    u8g2_Setup_ssd1306_i2c_128x64_noname_f(&u8g2, U8G2_R0, u8x8_HW_com_twi_nrf52832, u8g2_nrf_gpio_and_delay_twi_cb);    
    u8g2_SetI2CAddress(&u8g2, OLED_ADDR);
    u8g2_InitDisplay(&u8g2);
    u8g2_SetPowerSave(&u8g2,0);

}

void mui_u8g2_deinit(u8g2_t *p_u8g2) {
    u8g2_SetPowerSave(p_u8g2, 1);
}

void mui_u8g2_set_oled_contrast_level(uint8_t value) {
    //mui_t *p_mui = mui();
    //u8g2_SetContrast(&p_mui->u8g2, (value - 1) * (255.0 / 99.0));
}
`

solosky, 2024-11-08:
Sorry, I dont have any device to test your code..
I think you could hook signal pin to logic analyzer and try to grab some data to see if the MCU send the correct data out first.

Ahmadkorhani, 2024-11-08:
That's what I will do, I have a DSLogic logic analyzer. I will share any update. One last question, the RTT is normal ? Is it missing anything ?

Ahmadkorhani, 2024-11-08:
I have added I2C scanning + log after twi_init() and the system detect the address correctly which is 0x3C so i think the code should work correctly, then i found that it looks like the nrf52 freezes for some reason. the 32khz crystal is working fine. as I can't debug inside VSCode I can't find the source of the issue, every time I try to compile it takes forever to build. it rebuild everything everytime

Ahmadkorhani, 2024-11-09:
I think I should quit, I don't know what is happening. I started a new ble project in segger and added both spi flash and ssd1306. they are working great with my code.
Adding same code to this project is not working, I am building this code using GNU Arm Embedded Toolchain\10 2021.10 inside VSCode.

I don't know what could be the problem and I am sure now it is not coming from my modification. Maybe some VScode configuration is messing with the building output. 

If anyone managed to build it and flash it successfully please specify what compiler you used and if you did some tweaks or not to the make file

solosky, 2024-11-09:
here is a hint, use new free pins to connect SDA and SLC signal, please do not use same pin as SPI. 
The oled and SPI flash are shared same SPI bus in current implementation.

Ahmadkorhani, 2024-11-09:
of course I did that, remember I added a function to detect I2C address and it was working, after all I found that the processor hangs inside the while after the initialization. So it looks like something is wrong in the building process

Ahmadkorhani, 2024-11-17:
Finally it is working with I2C SSD1306 OLED display, it was a silly pointer typing problem. Now everything is working like a charm :)

closed 2024-11-17 by Ahmadkorhani