solosky/pixl.js #290: I did a software update and allmiibo doesn't turn on, is there a way to fix it?

Eliasarusi, 2024-11-16:


MayanKoyote, 2024-11-16:
https://github.com/solosky/pixl.js/blob/main/docs/en/02-Flash-Firmware.md
"Repair wrong firmware version" section

Kleei, 2025-02-19:
> I have a white LCD 'allmiibo' device from Aliexpress, just updated to latest firmware using: pixjs_ota_v461.zip from LCD version ZIP and https://thegecko.github.io/web-bluetooth-dfu/examples/web.html
> 
> Update was ok, but when device is started I see nothing, just a backlight. No I didn't use OLED firmware (but tried it and saw stripes, but reverted back)
> 
> What is it? please

Same here, looks like LCD 2.12.0 have issues with display contrast. Waiting for a fix.

Goozoon, 2025-02-19:
> > I have a white LCD 'allmiibo' device from Aliexpress, just updated to latest firmware using: pixjs_ota_v461.zip from LCD version ZIP and https://thegecko.github.io/web-bluetooth-dfu/examples/web.html
> > Update was ok, but when device is started I see nothing, just a backlight. No I didn't use OLED firmware (but tried it and saw stripes, but reverted back)
> > What is it? please
> 
> Same here, looks like LCD 2.12.0 have issues with display contrast. Waiting for a fix. Meanwhile you can downgrade back to 2.11.2

Do you think it is Contrast issue? i see literally nothing at any angles

Kleei, 2025-02-19:
> 
> Do you think it is Contrast issue? i see literally nothing at any angles

> New firmware 2.12.0 enabled contract menu for LCD device, you may update to 2.12.0 to adjust LCD contract manually. 
>  

 _Originally posted by @solosky in [#297](https://github.com/solosky/pixl.js/issues/297#issuecomment-2599631666)_

i think so.

Goozoon, 2025-02-19:
Back to 2.11.2 and it works, so probably yes.

Kleei, 2025-02-20:
> Back to 2.11.2 and it works, so probably yes.

check this !
https://github.com/solosky/pixl.js/issues/297#issuecomment-2671275816